google-deepmind/mujoco #162: Windows: LINK : fatal error LNK1181: cannot open input file '..\bin\mujoco210nogl.lib'

rohit-kumar-j, 2022-03-02:
Error when running `C:\mujoco-2.1.1\sample> make` in  _x64 Native Tools Command Prompt for VS Code 2022_:
(I am using the [windows makefile version](https://github.com/deepmind/mujoco/blob/main/sample/Makefile.windows) that is provided)

```
C:\mujoco-2.1.1\sample>make
cl /O2 /MT /EHsc /arch:AVX /I../include /Fe../bin/ testxml.cc                 ../bin/mujoco210nogl.lib
Microsoft (R) C/C++ Optimizing Compiler Version 19.31.31104 for x64
Copyright (C) Microsoft Corporation.  All rights reserved.

testxml.cc
Microsoft (R) Incremental Linker Version 14.31.31104.0
Copyright (C) Microsoft Corporation.  All rights reserved.

/out:../bin/testxml.exe
testxml.obj
../bin/mujoco210nogl.lib
LINK : fatal error LNK1181: cannot open input file '..\bin\mujoco210nogl.lib'
make: *** [makefile:4: all] Error 2
```

saran-t, 2022-03-02:
Hi there, the makefiles are out of date (we intended to replace them with CMakeLists but that's still on our todo list, sorry about that). For the time being, you should be able to proceed by replacing the filename `mujoco210nogl.lib` with the actual filenames from the downloaded package.

rohit-kumar-j, 2022-03-03:
Thank you for the quick reply! I didn't realize it was such a mundane error. I have moved past that one and now have a new error. It can't find `GLFW/glfw3.h`. It doesn't seem to be in lib, bin, or include either.
```
derivative.cc
Microsoft (R) Incremental Linker Version 14.31.31104.0
Copyright (C) Microsoft Corporation.  All rights reserved.

/out:../bin/derivative.exe
derivative.obj
../lib/mujoco_nogl.lib
cl /O2 /MT /EHsc /arch:AVX /I../include /Fe../bin/ basic.cc ../bin/glfw3.lib  ../lib/mujoco_nogl.lib
Microsoft (R) C/C++ Optimizing Compiler Version 19.31.31104 for x64
Copyright (C) Microsoft Corporation.  All rights reserved.

basic.cc
basic.cc(18): fatal error C1083: Cannot open include file: 'GLFW/glfw3.h': No such file or directory
make: *** [makefile:8: all] Error 2
```

saran-t, 2022-03-03:
We stopped shipping GLFW on Windows and macOS with our 2.1.1 release. You can grab a copy from https://github.com/glfw/glfw/releases

rohit-kumar-j, 2022-03-03:
I have looked at installing GLFW on windows. It is plagued with issues😓. This is the error from not being able to install it properly:

```
basic.cc
Microsoft (R) Incremental Linker Version 14.31.31104.0
Copyright (C) Microsoft Corporation.  All rights reserved.

/out:../bin/basic.exe
basic.obj
../bin/glfw3.lib
../lib/mujoco_nogl.lib
basic.obj : error LNK2019: unresolved external symbol __imp_mjr_defaultContext referenced in function main
basic.obj : error LNK2019: unresolved external symbol __imp_mjr_makeContext referenced in function main
basic.obj : error LNK2019: unresolved external symbol __imp_mjr_freeContext referenced in function main
basic.obj : error LNK2019: unresolved external symbol __imp_mjr_render referenced in function main
..\bin\basic.exe : fatal error LNK1120: 4 unresolved externals
make: *** [makefile:8: all] Error 2
```
Do you have simplified instructions for compiling with GLFW on windows?

rohit-kumar-j, 2022-03-03:
Thanks! this helps. No errors with the build. I have an error with `mj_loadXML` while running the `.exe` file. Perhaps I should start another issue.
```
Error=XML_ERROR_FILE_NOT_FOUND ErrorID=3 (0x3) Line number=0: filename=C:/mujoc0-2.1.1/Custom_projects/test_1/ball.xml
```

Inside the `hello.c` file:
```
m = mj_loadXML("C:/mujoc0-2.1.1/Custom_projects/test_1/ball.xml", NULL, error, 1000);
```

closed 2022-03-03 by rohit-kumar-j