google-deepmind/mujoco #163: Error: repeated name in mesh array while converting URDF

alaurenzi, 2022-03-03:
Hello everyone, 
I am trying to convert a URDF to XML using the `compile` program. I am getting this error though:
```
Error: repeated name in mesh array, position 31
Object name = hip-pitch1, id = 31
```
which, I guess, originates from the fact that the same mesh is used in multiple places. Indeed, the robot has four "hip pitch" links, each one containing both visual and collision description. Collision meshes are specified as `<mesh-dir>/collision/hip-pitch.stl`, whereas visual meshes as `<mesh-dir>/hip-pitch.stl`, and the `scale` attribute is used to flip the meshes left-to-right. 

I think this is common practice to avoid mesh duplication (the robot has four identical legs!) How should I deal with this situation?

Thanks!

yuvaltassa, 2022-03-03:
This could be considered a bug. MuJoCo currently strips the path from filenames internally leading to these unwanted collisions. We hope to fix this soon. In the meantime, please rename your meshes.

vicltz, 2023-06-20:
Is this bug fixed ? I have the same error on 2.35 with no other obvious reason than the same mesh used multiple times

yuvaltassa, 2023-11-23:
Does setting the [strippath](https://mujoco.readthedocs.io/en/latest/XMLreference.html#compiler-strippath) attribute to "false" not solve this?

yuvaltassa, 2024-01-08:
Closing for now.

closed 2024-01-08 by yuvaltassa

yuvaltassa, 2024-09-05:
I believe this was fixed in d12fe594d98ee0500fa6bb7513f3faf3d9f7488a, could you check and confirm?