google-deepmind/mujoco #168: WARNING: Nan, Inf or huge value in QACC at DOF xx. The simulation is unstable. Time = <sim time value>.

rohit-kumar-j, 2022-03-05:
I have a rather large model and I am using the [simulate](https://github.com/deepmind/mujoco/blob/main/sample/simulate.cc) example to build the robot. It has many kinematic loops. However, I am yet to create the loops. When I simulate the motion(all contacts disabled), there is a warning produced and it resets the simulation. (This keeps happening in a loop)

Is there a way to disable this?

```
C:\mujoco200_win64\bin>simulate.exe ../rojects/projname/robot.xml

MuJoCo Pro version 2.00
WARNING: Nan, Inf or huge value in QACC at DOF 35. The simulation is unstable. Time = 2.7620.
```
[image]

MotorCityCobra, 2022-03-05:
I'm not qualified to fully answer this, but it means you've written some part of the code to be unstable. Does it happen right away with the model or does it happen when you move the joints?  

You might get a better response if you add the meshes and the XML file.

Lately the devs have been good at answering questions in here.

yuvaltassa, 2022-03-05:
Various issues can cause instability.  
Try adding some damping or armature to your joints and check if you have any self-collisions at the initial pose.
If you post you model here we can help you diagnose.

closed 2022-03-05 by yuvaltassa

rohit-kumar-j, 2022-03-05:
Thanks for the suggestions! (I forgot to mention this before, but I'm using [mujoco200 win64](https://www.roboti.us/download.html))

Yes, the convex hulls of the parts intersect in their initial pose. Although all collision is disabled. (see below for XML)

There is no damping or armature in the joints. It consists of hydraulic cylinders and hence I need to create a layer that takes in forces and velocities from the simulation and convert it to flowrate and pressure at each joint actuator. 
So, I might need to specify parameters like joint friction.

How it looks:
(When I hit **run**, it runs okay for a few seconds, then resets )
[image]

 XML and meshes: [LINK](https://drive.google.com/file/d/1ue8Zuk4z6sYOjvohLZoVnkyDLYJOFdft/view?usp=sharing) (Restricted sharing. Change the path in XML files for accessing the mesh files)

yuvaltassa, 2022-03-05:
You system is basically a chaotic energy conserving system. This kind of thing is bound to de-stabilize with Euler.
Switch the integrator to RK4. Or add joint friction :)

rohit-kumar-j, 2022-03-05:
Thanks! Changing to RK4 solved the problem. I was wondering what would the difference be in changing from Euler to RK4?

MyelinsheathXD, 2022-05-30:
This resetting situation is coming from large contact or joint constraint violation. That would be great if  a solver fixes this violation issue within cheap Euler stepping method for real time apps.

ryanchenstats, 2023-10-26:
> Thanks! Changing to RK4 solved the problem. I was wondering what would the difference be in changing from Euler to RK4?

runge kutta generally has better convergence rates than euler since euler is makes a single derivative approximation while RK4 makes 4 approximations on a given interval.