google-deepmind/mujoco #169: Model made of one joint rotates 360 degrees even though I have limited="true" set

MotorCityCobra, 2022-03-05:
I have a model consisting of two meshes with a joint connecting. The model should only move within the range set at `range="-3.0 0"` but it rotates more than 360 degrees through the meshes in an endless loop.  

I was directed to the [Activation dynamics[(https://mujoco.readthedocs.io/en/latest/computation.html#numerical-integration) section of the docs before. But it's not clear to me from the docs what part of this math is the gain, dynamics, and bias. I would like to learn, but I am lost. In the Simulation sample file I can click on the joint actuator and make the joint move a specific angle and at a specific speed. I would like to know how to do this with the gain, dynamics, and bias with functions in the C++ file. Specific movements and speeds.

[The XML model and meshes can be downloaded here](https://drive.google.com/drive/folders/1eneA9j36rqCaI63i2hlmMQv-or75-iZw?usp=sharing)

The relevant parts of the main function I modified of the simulate.cc file.
Thank you.

```
int dyn;
int bias;
int gain;

mjtNum gain_callback(const mjModel *m, const mjData *d, int id_01)
{
    return gain;
}
mjtNum dyn_callback(const mjModel *m, const mjData *d, int id_01)
{
    return dyn;
}
mjtNum bias_callback(const mjModel *m, const mjData *d, int id_01)
{
    return bias;
}

// void myController(const mjModel* mArg, mjData* dArg) {
// //Some useful code here
// dArg->act[0] = -100.9;
// }

// run event loop
int main(int argc, const char** argv)
{
    mjcb_act_gain = gain_callback;

    // initialize everything
    init();

    // request loadmodel if file given (otherwise drag-and-drop)
    if( argc>1 )
    {
        mju_strncpy(filename, argv[1], 1000);
        settings.loadrequest = 2;
    }

    // start simulation thread
    std::thread simthread(simulate);

    // event loop
    while( !glfwWindowShouldClose(window) && !settings.exitrequest )
    {
        // start exclusive access (block simulation thread)
        mtx.lock();

        // load model (not on first pass, to show "loading" label)
        if( settings.loadrequest==1 )
            loadmodel();
        else if( settings.loadrequest>1 )
            settings.loadrequest = 1;

        // handle events (calls all callbacks)
        glfwPollEvents();

        // prepare to render
        prepare();

        // end exclusive access (allow simulation thread to run)
        mtx.unlock();

        // render while simulation is running
        render(window);
        if (m)
        {

            if ((int)((int)d->time % (int)4) == (int)0 || (int)((int)d->time % (int)4) == (int)1)
            {
                dyn = 10;
                bias = 0;
                gain = 12.4;
            } else
            {
                dyn = 10;
                bias = 0;
                gain = 12.4;
            }
        }
    }
```

yuvaltassa, 2022-03-05:
There are multiple issues with your model (which I had to edit quite a bit to get to load, the meshes have bad names and paths). For example the hinge joint is nowhere near where it should be. In any case, after uncommenting the native actuator, the joint limits you specified are maintained. Remember that joint limits in MuJoCo, like all constraints, are soft, so some small violation is to be expected.

closed 2022-03-05 by yuvaltassa

MotorCityCobra, 2022-03-06:
Hmmm. When I open the XML file it loads and rotates at the joint,