google-deepmind/mujoco #172: how can i get some example to simulate closed loop mechanism, [four-bar linkage]

chinahuangyong, 2022-03-09:


haoranlab, 2022-03-12:
As far as I know, there's no native support for closed loop mechanism in mujoco, but you can try adding some 'equality' object to constraint the kinematic chain.

yuvaltassa, 2022-03-14:
Indeed, 4-bar linkages should be implemented using equality constraints. We hope to have some example models soon showing how to do this.

closed 2022-03-14 by yuvaltassa

rohit-kumar-j, 2022-05-12:
@chinahuangyong, here is a simple example. You can use this in the default simulate viewer.
(Here I used [mujoco-python-viewer](https://github.com/rohanpsingh/mujoco-python-viewer/blob/main/examples/sample.py) `pip install mujoco-python-viewer`)

<details>
    <summary>Example video (If the video get stuck, scrub a little forward)</summary>

https://user-images.githubusercontent.com/37873142/168159575-70b1b520-8758-4824-8a15-1e176667e95d.mp4

</details>

<details>
    <summary>Example xml file</summary>

```xml
<mujoco>
    <option gravity="0 0 0" timestep="0.002" />
    <asset>
        <material name="blue_" rgba="0 0 1 1" />
        <material name="green" rgba="0 1 0 1" />
        <material name="red__" rgba="1 0 0 1" />
        <material name="white" rgba="1 1 1 1" />
    </asset>
    <worldbody>
        <geom type="plane" size="10 10 0.1" pos="0 0 -2" rgba=".9 0 0 1" />
        <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1" />
        <body name="link_1" pos="0 0 0">
            <geom type="cylinder" size=".2    2" pos="0 0 2" euler="0 0 0" material="red__" />
            <geom type="cylinder" size=".25 .25" pos="0 0 4" euler="0 90 0" material="red__" />
            <geom type="cylinder" size=".25 .25" pos="0 0 0" euler="0 90 0" material="red__" />
            <body name="link_2" pos="0.5 0 0" euler="0 0 0">
                <joint name="hinge_1" pos="0 0 0" axis="1 0 0" />
                <geom type="cylinder" size=".2    2" pos="0 2 0" euler="90 0 0" material="blue_" />
                <geom type="cylinder" size=".25 .25" pos="0 4 0" euler="0 90 0" material="blue_" />
                <geom type="cylinder" size=".25 .25" pos="0 0 0" euler="0 90 0" material="blue_" />
                <body name="link_3" pos="-0.5 4 0" euler="0 0 0">
                    <joint name="hinge_2" pos="0 0 0" axis="1 0 0" />
                    <geom type="cylinder" size=".2    2" pos="0 0 2" euler="0 0 0" material="green" />
                    <geom type="cylinder" size=".25 .25" pos="0 0 0" euler="0 90 0" material="green" />
                    <geom type="cylinder" size=".25 .25" pos="0 0 4" euler="0 90 0" material="green" />
                    <body name="link_4" pos="0.5 0 4" euler="0 0 0">
                        <joint name="hinge_3" pos="0 0 0" axis="1 0 0" />
                        <geom type="cylinder" size=".2    2" pos="0 -2 0" euler="90 0 0" material="white" />
                        <geom type="cylinder" size=".25 .25" pos="0 0 0" euler="0 90 0" material="white" />
                        <geom type="cylinder" size=".25 .25" pos="0 -4 0" euler="0 90 0" material="white" />
                    </body>
                </body>
            </body>
        </body>
    </worldbody>
    <equality>
        <connect name="kinematic_link" active="true" body1="link_1" body2="link_4" anchor="0 0 4" />
    </equality>
</mujoco>
```
</details>

<details>
    <summary>Example python script to run the sim</summary>

```python3
import mujoco
import mujoco_viewer

model = mujoco.MjModel.from_xml_path("path_to_4_bar.xml")
data = mujoco.MjData(model)

# create the viewer object
viewer = mujoco_viewer.MujocoViewer(model, data)

# simulate and render
for _ in range(100000):
    mujoco.mj_step(model, data)
    viewer.render()

# close
viewer.close()
```

</details>

kbichave, 2023-11-01:
Hello!

I encountered this example while searching for a solution to the issue I'm experiencing with a 4-bar linkage. In my specific situation, the joint created using equality constraints proves to be unstable and frequently disengages. Additionally, I'm encountering difficulties in configuring the joint in my physical system to behave as a hinge, and I'm unsure how to establish closed-loop dynamics for it. Any assistance in this regard would be greatly appreciated.

The problem of joint disengagement is evident in the simulation provided below. In this simulation, I've connected a motor to one of the joints and moved it rapidly, clearly demonstrating the disengagement of the equality ball joint. This disengagement issue presents a challenge in my actual, unmentioned simulation.

Thank you in advance for any assistance.

`<mujoco>
    <option timestep="0.0002" />
    <actuator>
        <motor gear="10000" joint="hinge_1" name="actuator_motor" ctrlrange="-300 300" ctrllimited="true"/>
    </actuator>
    <asset>
        <material name="blue_" rgba="0 0 1 1" />
        <material name="green" rgba="0 1 0 1" />
        <material name="red__" rgba="1 0 0 1" />
        <material name="white" rgba="1 1 1 1" />
    </asset>
    <worldbody>
        <geom type="plane" size="10 10 0.1" pos="0 0 -5" rgba=".9 0 0 1" />
        <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1" />
        <body name="link_1" pos="0 0 0">
            <geom type="cylinder" size=".2    2" pos="0 0 2" euler="0 0 0" material="red__" />
            <geom type="cylinder" size=".25 .25" pos="0 0 4" euler="0 90 0" material="red__" />
            <geom type="cylinder" size=".25 .25" pos="0 0 0" euler="0 90 0" material="red__" />
            <body name="link_2" pos="0.5 0 0" euler="0 0 0">
                <joint name="hinge_1" pos="0 0 0" axis="1 0 0" />
                <geom type="cylinder" size=".2    2" pos="0 2 0" euler="90 0 0" material="blue_" />
                <geom type="cylinder" size=".25 .25" pos="0 4 0" euler="0 90 0" material="blue_" />
                <geom type="cylinder" size=".25 .25" pos="0 0 0" euler="0 90 0" material="blue_" />
                <body name="link_3" pos="-0.5 4 0" euler="0 0 0">
                    <joint name="hinge_2" pos="0 0 0" axis="1 0 0" />
                    <geom type="cylinder" size=".2    2" pos="0 0 2" euler="0 0 0" material="green" />
                    <geom type="cylinder" size=".25 .25" pos="0 0 0" euler="0 90 0" material="green" />
                    <geom type="cylinder" size=".25 .25" pos="0 0 4" euler="0 90 0" material="green" />
                    <body name="link_4" pos="0.5 0 4" euler="0 0 0">
                        <joint name="hinge_3" pos="0 0 0" axis="1 0 0" />
                        <geom type="cylinder" size=".2    2" pos="0 -2 0" euler="90 0 0" material="white" />
                        <geom type="cylinder" size=".25 .25" pos="0 0 0" euler="0 90 0" material="white" />
                        <geom type="cylinder" size=".25 .25" pos="0 -4 0" euler="0 90 0" material="white" />
                    </body>
                </body>
            </body>
        </body>
    </worldbody>
    <equality>
        <connect name="kinematic_link" active="true" body1="link_1" body2="link_4" anchor="0 0 4" />
    </equality>
    <contact>
        <exclude body1="link_1" body2="link_2"/>
        <exclude body1="link_1" body2="link_3"/>
        <exclude body1="link_1" body2="link_4"/>
        <exclude body1="link_2" body2="link_3"/>
        <exclude body1="link_2" body2="link_4"/>
        <exclude body1="link_3" body2="link_4"/>
    </contact>
</mujoco>`

yuvaltassa, 2023-11-02:
Try modifying the [solver parameters](https://mujoco.readthedocs.io/en/latest/modeling.html#solver-parameters) of your equality constraint. 

Duplicate of #906