google-deepmind/mujoco #173: unexpected behavior in simulation of rope attached to a robot

xibeisiber, 2022-03-09:
Hi,
I want to simulate a rope attached to the end of iiwa. But something unexpected happened when I tried to let one joint of iiwa move.

The screencast is [here](https://github.com/xibeisiber/pics/blob/master/aa/screencast.mp4).
The xml file and script are in [https://github.com/xibeisiber/pics/tree/master/aa](https://github.com/xibeisiber/pics/tree/master/aa).

I'm not sure what the problem is.
Thanks for any help!

xibeisiber, 2022-03-10:
solved by using qfrc_applied instead of set_joint_qpos...

closed 2022-03-10 by xibeisiber

jiyooonp, 2023-03-29:
Hello, 
I am trying to simulate a ball on a string on my manipulator too, but I am not sure how to start making the XML file. Could you share your XML file if possible? 
Thank you so much!

jiyooonp, 2023-04-02:
@yuvaltassa Thank you so much!