google-deepmind/mujoco #175: How would you simulate a stepper motor, or at least get a hinge joint to move a specific length, from one angle to another?

MotorCityCobra, 2022-03-09:
I would like to simulate a stepper motor.  
Stepper motors precisely move from position to another angle. Stepper motors also tend to hold the positions they are in.  

If simulating a stepper motor is too much to ask I would like to know how to move a hinge joint only by a certain angle using the mjcb callbacks. For example, if I could get an actuator to move a joint to only move 20 degrees, or if I could only get an actuator to apply force for 2 seconds and then stop. 

**My Attempt**
I modified the `simulate.cc` file to move only for a small duration using `mjcb_act_dyn` and `mjcb_act_gain` callbacks.  
I also set the stiffness of the joint to `2` in the XML model file.  
I expected the joint to just move a little bit. `100` steps, if you will.  
Instead it opened the joint all the way.  

[Here is the link to the XML and the meshes](https://drive.google.com/drive/folders/1ICQH2QhD0dsLQF8HE07sYVT9U45eZbAp?usp=sharing)

[Here is the modified simulate.cc C++ file](https://drive.google.com/file/d/10Ew7xA8--WAAZTVxKix0uZLUsFz2RnT-/view?usp=sharing)

Below I have included the changes I made to the simulate.cc file

```
int dyn;
int bias;
int gain;

mjtNum gain_callback(const mjModel *m, const mjData *d, int id_01)
{
    return gain;
}
mjtNum dyn_callback(const mjModel *m, const mjData *d, int id_01)
{
    return dyn;
}
mjtNum bias_callback(const mjModel *m, const mjData *d, int id_01)
{
    return bias;
}

int flip;

int first_count;
int second_count;

// run event loop
int main(int argc, const char** argv)
{
    mjcb_act_gain = gain_callback;

    first_count = 100;

    // initialize everything
    init();

    // request loadmodel if file given (otherwise drag-and-drop)
    if( argc>1 )
    {
        mju_strncpy(filename, argv[1], 1000);
        settings.loadrequest = 2;
    }

    // start simulation thread
    std::thread simthread(simulate);

    // event loop
    while( !glfwWindowShouldClose(window) && !settings.exitrequest )
    {
        // start exclusive access (block simulation thread)
        mtx.lock();

        // load model (not on first pass, to show "loading" label)
        if( settings.loadrequest==1 )
            loadmodel();
        else if( settings.loadrequest>1 )
            settings.loadrequest = 1;

        // handle events (calls all callbacks)
        glfwPollEvents();

        // prepare to render
        prepare();

        // end exclusive access (allow simulation thread to run)
        mtx.unlock();

        // render while simulation is running
        render(window);
        if (m)
        {
            mjcb_act_dyn = dyn_callback; 
            mjcb_act_bias = bias_callback;

            if ((int)((int)(d->time*10) % (int)1) == (int)0 && ((int)d->time > (int)2))
            {
                if (first_count > 0)
                {
                    first_count--;
                    std::cout << first_count << "\n";
                    // bias = -1;
                    gain = -1;
                    dyn = 2.9;

                }
            }
        }
    }

```

yuvaltassa, 2022-03-09:
You don't need the callbacks. Just use a high gain [position actuator](https://mujoco.readthedocs.io/en/latest/XMLreference.html#actuator-position) on a joint with enough damping.

closed 2022-03-09 by yuvaltassa

MotorCityCobra, 2022-03-10:
> You don't need the callbacks.

@yuvaltassa 
If I don't use callbacks then how would I control the actuator from the C++ file?

yuvaltassa, 2022-03-10:
`data->ctrl[i] = something;`

MotorCityCobra, 2022-03-10:
@yuvaltassa 
Thanks.
Though I commented out the callbacks, used d->ctrl, used a motor joint, added damping, but the joint still opens all the way. It'll open slower or faster depending on how much damping I set for the joint. It opens all the way though I have a loop apply the `d->ctrl[0] to the joint for a short amount of time.

MotorCityCobra, 2022-03-10:
Is there a way to send a conflicting commands to the joint to get it to stay in place instead of damping? I know this doesn't work because I tried it, but something like `d->ctrl[0] = -1` and `d->ctrl[0] = 1`?