google-deepmind/mujoco #179: How to comprehend gear attribute in actuator(motor)?

Coder-RG, 2022-03-10:
I am trying to figure out how to assign the `gear` attribute to *motor* in my quadrotor model. I found some code online where the gear values were set as `0 0 1 0 0 0.1`. As I understand this, the first three values specify the force along the principal axis and the next three specify the torque. However, when I changed the torque value from **0.1** to **1**. Then it did not show any difference.

Here is the actuator code I am using. The code works well but changing the last value from 0.1 to 1 does not have any effect, at least not noticeable.

```
    <actuator>
        <motor ctrllimited="true" ctrlrange="0.0 1.0" gear="0 0. 1. 0. 0. -1" site="motor1"/>
        <motor ctrllimited="true" ctrlrange="0.0 1.0" gear="0 0. 1. 0. 0. 1" site="motor2"/>
        <motor ctrllimited="true" ctrlrange="0.0 1.0" gear="0 0. 1. 0. 0. -1" site="motor3"/>
        <motor ctrllimited="true" ctrlrange="0.0 1.0" gear="0 0. 1. 0. 0. 1" site="motor4"/>
    </actuator>
```

MotorCityCobra, 2022-03-10:
What is the scale of your model?  
Try 100, 1,000, 10,000, 100,000 etc.

Did you make the meshes yourself? If you make your units MKS then gear="1" will be 1 Newton Meter.

MotorCityCobra, 2022-03-10:
What happens when you use just one value for the gear?
i.e. `gear="100"`

Coder-RG, 2022-03-10:
> What happens when you use just one value for the gear? i.e. `gear="100"`

It went completely sideways, i.e. in the +x direction. At least earlier it was behaving like a quadrotor.😂

MotorCityCobra, 2022-03-10:
In the simulator screen click rendering and click Actuator to see where your actuator is.

You might want to do this with the example models to learn how to put the joint exactly where it needs to be. You can make meshes in Solidworks and get the x, y, z from them. It takes work to learn and get right.

[image]

Coder-RG, 2022-03-10:
I did! But I don't really have a problem with the placement. Rather, the actual gear values. Currently, the rotor works as it is supposed to. However, I am trying to understand what each value means.

MotorCityCobra, 2022-03-10:
You say you don't have a problem with the placement and you also said it went "completely sideways".  
When you render the Actuator does your joint look like that? Is your joint in the center of the Actuator cone / cylinder?  

Like I said, you'll probably want to wait for @yuvaltassa , the developer who usually answers the questions in here.

Coder-RG, 2022-03-10:
What I meant was after using a single value for gear as you said, the model did not perform the way it was expected. This means it is expecting other values in the `gear` attribute as mentioned in the doc.

yuvaltassa, 2022-03-15:
Did you figure this out? The 6th number in the gear vector should indeed change the torque value along the z direction. Perhaps it is being cancelled out by the other motors applying negative torque? 

If you still need help, please post a full loadable model and we will inspect.

closed 2022-03-15 by yuvaltassa