google-deepmind/mujoco #182: Different performance between muscle and motor actuators

haoranlab, 2022-03-12:
Hi, I'm trying to build a simple  mechanism like: 
ground ---> hinge joint1 around y-axis ---> body1 ---> hinge joint2 around y-axis ---> body2

During the experiment, I find the behavior of the system is different when using different actuators to drive joint2.

When using the motor to drive joint2, the torque applied to joint2 will not cause joint1 rotating. However, when using a pair of muscle actuators to drive joint2, joint1 will rotate following the movement of joint2. I'm confused about this strange phenomenon and I wonder if there're any methods to drive joint2 with muscles while avoiding affecting joint1?

yuvaltassa, 2022-03-14:
What you are describing is indeed odd, but please remember muscles are not supposed to be attached to joints, only to tendons.

closed 2022-03-14 by yuvaltassa

haoranlab, 2022-03-14:
> What you are describing is indeed odd, but please remember muscles are not supposed to be attached to joints, only to tendons.

Thanks for your valuable advice! What dose it mean to attach muscle to joints? I did attach muscles on tendons and the xml file is as follow:
`<mujoco model="demo">
    <compiler angle="degree" meshdir="../geometry"/>
    <option timestep="0.005" iterations="50" solver="Newton" tolerance="1e-10" collision="predefined"/>
    <size njmax="200" nconmax="50" nstack="3000"/>

    <visual>
        <rgba haze=".3 .3 .3 1"/>
    </visual>

    <default>
        <joint limited="true" range="-90 90" damping="1.0"/>

        <default class="muscle_site">
            <site type="sphere" rgba="0 0 1 1"></site>
        </default>

        <default class="muscle_tendon">
            <tendon width="0.002" rgba="1 0 0 1" limited="true" range="0 1"></tendon>
        </default>

        <default class="muscle_actuator">
            <muscle force="150" ctrlrange="0 1"></muscle>
        </default>
    </default>

    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.44 0.85 0.56" rgb2="0.46 0.87 0.58" width="32" height="32"/>
    </asset>

    <worldbody>
        <geom name="ground" type="plane" pos="0 0 0" size="0.5 0.5 2" rgba="0 0 0.5 0.5"/>
        <site name="m1e" pos="0.05 -0.23 0.185" class="muscle_site"/>
        <site name="m2e" pos="-0.05 -0.23 0.185" class="muscle_site"/>
        <body name="base" pos="0 -0.08 0.385">
            <geom name="base_center" type="box" size="0.025 0.011 0.025" rgba="1 0 0 0.8"/>
            <geom name="base_axis_y" type="cylinder" fromto="0 -0.1 0 0 0.1 0" size="0.01"/>
            <geom name="base_axis_x" type="cylinder" fromto="-0.1 0 0 0.1 0 0" size="0.01"/>

            <body name="child1" pos="0 -0.1 0">
                <geom name="r1" type="box" size="0.05 0.01 0.05" rgba="0 1 1 0.8"/>
                <joint name="j1" type="hinge" pos="0 0 0" axis="0 -1 0"/>
                <body name="child2" pos="0 -0.05 0">
                    <geom name="r3" type="box" pos="0 0 0" size="0.05 0.01 0.05" rgba="0 1 1 0.8"/>
                    <joint name="j2" type="hinge" pos="0 0 0" axis="0 -1 0"/>
                    <site name="m1b" pos="0.05 0 0" class="muscle_site"/>
                    <site name="m2b" pos="-0.05 0 0" class="muscle_site"/>
                </body>
            </body>
        </body>
    </worldbody>

    <tendon>
        <spatial name="m1" class="muscle_tendon">
            <site site="m1b"/>
            <site site="m1e"/>
        </spatial>

        <spatial name="m2" class="muscle_tendon">
            <site site="m2b"/>
            <site site="m2e"/>
        </spatial>
    </tendon>

    <actuator>
        <motor name="motor" joint="j2" ctrlrange="-1 1"/>
        <muscle name="m1" class="muscle_actuator" tendon="m1"/>
        <muscle name="m2" class="muscle_actuator" tendon="m2"/>
    </actuator>
</mujoco>`

yuvaltassa, 2022-03-14:
Hi. Thanks for the model, that was very helpful (though somehow the `<mujoco>` clause was deleted? I added it back).

This is an interesting situation. What is happening is the following. Your muscles are applying external forces/torques to the second body. These are propagating to the first body via the damping which you have set to 1.0 and both bodies turn. However the motor is acting directly on the joint, it produces _internal_ rather than _external_ torques. This means that if `f` is being applied to to `child2`, `-f` is being applied to `child1` via Newton's 3rd law. So what your motor is trying to do is actually rotate `child1` in the _opposite_ direction. This is harder to do, since it must overcome the dampings on both joints, rather than just one joint, as when rotating `child1`. If you set your damping to some small value (say 0.01), you will see your motor spins both bodies, but in opposite directions.

haoranlab, 2022-03-14:
Thanks for your help! It would be extremely hard for me to figure this out myself.