google-deepmind/mujoco #184: How can I change the kv (damping) of a joint to a higher value and how can I calculate it to hold up to 0.5 NM?

MotorCityCobra, 2022-03-14:
I'm using a function to change the velocity on a joint to force it to hold still between movements. I use a function below to change the kv with `m->actuator` gain and bias, but it becomes unstable if I set the value near where I want it.  

If I set the kv to anything higher than `1` I get the error `WARNING: Nan, Inf or huge value in QACC at DOF 0. The simulation is unstable. Time = 4.4530.`

```
    <actuator>
        <motor name="hinge" gear="0.156766" joint="hinge" />
        <position name="position_servo" joint="hinge" kp="0" />
        <velocity name="velocity_servo" joint="hinge" kv="0" />
    </actuator>
```

```
void set_velocity_servo(const mjModel* m, int actuator_no, double kv)
{
  m->actuator_gainprm[10*actuator_no+0] = kv;
  m->actuator_biasprm[10*actuator_no+2] = -kv;
}
```

```
double kv = 0.3;
set_velocity_servo(m, 2, kv);
```

Am I on the right track? Is this the best way to essentially put the breaks on a joint and keep it rigid through the API? I am trying to simulate a NEMA 17 stepper motor, which has a holding torque of 0.5 Newton Meters.  
I was considering a different approach of getting a reading of where the joint was after it was done moving and set a PID control to go to, in other words, stay at that position. Though what I really want is something that takes 0.5 Newton Meters to move a step.

yuvaltassa, 2022-03-14:
Try reducing the timestep and/or switching to the RK4 integrator

closed 2022-03-14 by yuvaltassa

MotorCityCobra, 2022-03-15:
Can I switch to RK4 with something in the `option` tag in the XML?  
I know I'm always asking question after question after question, but I searched the repo and don't see an example of it being implemented.  

How much can I reduce the timestep without causing problems? I reduced it one hundred times and increased the damping one hundred times. The joint still moves slowly. If I have a robot doing a task I can't have its knees bending a micrometer.

yuvaltassa, 2022-03-15:
`<option integrator="RK4"/>`
You searched the repo? You know MuJoCo has [documentation](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option), right?

Reduce the timestep as much as you like, there are no side effects, other than having to take more steps.

MotorCityCobra, 2022-03-15:
> You know MuJoCo has [documentation](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option), right?

lol, yes.  
I thought I might see an example of how to implement it in the XML examples. When I searched 'RK4' in the docs I saw three search results and thought it was only mentioned three times in the docs. Though it is on the XML Reference page twice.  

I would still like to humbly make a request to have 'holding torque' added as a feature of the simulator. To have a joint rigid when not in an actuating state that a specific amount of external force will apply x amount of torque to rotate.

yuvaltassa, 2022-03-15:
What would you say to a new ("dashpot"?) actuator type that would let you actively set the joint damping using a ctrl value? That's a general enough feature that would let you implement what you want, correct?

MotorCityCobra, 2022-03-15:
> What would you say to a new ("dashpot"?) actuator type that would let you actively set the joint damping using a ctrl value?

I think I've applied as much damping as I can with the function below and the joints still move slowly. If I have a robot with a few joints slowly moving I won't able to accurately train a model to deploy in a real life robot. A real life robot won't have joints that are slowly turning. This dashpot would be more damping?  
 
Is there a way to essentially build a brake for a joint that can be overcome by a force? A brake that holds it in place totally unless a force threshold is met? That would do it.  

I'd like joints that hold completely still unless an amount of force can overcome a specific holding torque. Like I mentioned, the NEMA 17 stepper motor takes 0.5 NM of torque to turn a step from external force.  

At this point I'm ready to set a PD controller to hold the position of a joint, and maybe have it move if a force about 0.5 NM is read by a sensor.
```
void set_velocity_servo(const mjModel* m, int actuator_no, double kv)
{
  m->actuator_gainprm[10*actuator_no+0] = kv;
  m->actuator_biasprm[10*actuator_no+2] = -kv;
}
```

yuvaltassa, 2022-03-15:
Sounds like you want [dry joint friction](https://mujoco.readthedocs.io/en/latest/computation.html?highlight=frictionloss#friction-loss) rather than damping? Damping can't bring velocity to zero, it's proportional to velocity.

MotorCityCobra, 2022-03-16:
I saw this as an option for joint in the XML.  
I don't see it as an option for the joint through the API. I need to do this through the API. 
 
Maybe the feature goes by other terms in the API?

[image]

yuvaltassa, 2022-03-16:
[dof_frictionloss](https://github.com/deepmind/mujoco/blob/main/include/mjmodel.h#L640) is what you are looking for.

MotorCityCobra, 2022-03-16:
> [dof_frictionloss](https://github.com/deepmind/mujoco/blob/main/include/mjmodel.h#L640) is what you are looking for.

I couldn't get that to halt the joint.
I tried using it before. Since I don't quite understand it, I just tried again with all of these variations...  
`m->dof_frictionloss[0] = 10;`
`m->dof_frictionloss[0] = -10;`
`m->dof_frictionloss[1] = 10;`
`m->dof_frictionloss[1] = -10;`
`m->dof_frictionloss[2] = 10;`
`m->dof_frictionloss[2] = -10;`
`m->dof_frictionloss[3] = 10;`
`m->dof_frictionloss[3] = -10;`  

I used 1000, and -1000 as well.  

I tried another attempt... I set `d->ctrl[0] = my_desired_qpos` and it does not stay at the position. Shouldn't it go to / stay in the position?

Some info on my model, which is in more detail in [this other GitHub question I posted here](https://github.com/deepmind/mujoco/issues/175#issue-1164472905)
[image]