google-deepmind/mujoco #186: Break in `mjlib.mj_jacSite` to `mujoco.mj_jacSite` switch

kevinzakka, 2022-03-15:
Hi,

My inverse kinematics code is currently breaking after the upgrade. Following the [migration guide](https://github.com/deepmind/dm_control/blob/master/migration_guide_1.0.md), I replaced my code as follows:

```python
# Before
mjlib.mj_jacSite(
    model.ptr,
    data.ptr,
    jacobian_position,
    jacobian_rotation,
    object_id,
)

# After
mujoco.mj_jacSite(
    model.ptr,
    data.ptr,
    jacobian_position,
    jacobian_rotation,
    object_id,
)
```

The error I get is:

> TypeError: mj_jacSite(): incompatible function arguments. The following argument types are supported:
    1. (m: mujoco._structs.MjModel, d: mujoco._structs.MjData, jacp: Optional[numpy.ndarray[numpy.float64[m, n], flags.writeable, flags.c_contiguous]], jacr: Optional[numpy.ndarray[numpy.float64[m, n], flags.writeable, flags.c_contiguous]], site: int) -> None

I double checked all types as well as writability of arrays so unclear why the error is being thrown.

kevinzakka, 2022-03-15:
Also just noticed getting errors with `mj_integratePos` as well. So not specific to `mj_jacSite`!

saran-t, 2022-03-15:
The error message should've said what pybind thinks your argument types are as well, can you post that too?

saran-t, 2022-03-15:
You haven't updated `dm_control` :)