google-deepmind/mujoco #187: How can i install the callback "mj_passive"?

Uptomylimit, 2022-03-16:
I want to do some fluid dynamics computation, so i need to install the callback "mj_passive",but i don't kow how to install the callback.
Thanks you anyway!

saran-t, 2022-03-16:
You need to implement your custom passive force function that is of the [`mjfGeneric`](https://mujoco.readthedocs.io/en/latest/APIreference.html?highlight=mjcb_passive#mjfgeneric) type (aka `void (*)(const mjModel*, mjData*)`), and assign it to the [`mjcb_passive`](https://mujoco.readthedocs.io/en/latest/APIreference.html?highlight=mjcb_passive#mjcb-passive) global function pointer.

If you're in Python, you can also write a Python function and pass it to `mujoco.set_mjcb_passive`, however note that this will potentially significantly slow down MuJoCo, so is probably only good for prototyping.

closed 2022-03-16 by saran-t

Uptomylimit, 2022-03-17:
Do you mean this function in mujoco.core?
`
def set_callback(name, new_callback=None):
  """Sets a user-defined callback function to modify MuJoCo's behavior.

  Callback functions should have the following signature:
    func(const_mjmodel_ptr, mjdata_ptr) -> None

  Args:
    name: Name of the callback to set. Must be a field in
      `functions.function_pointers`.
    new_callback: The new callback. This can be one of the following:
      * A Python callable
      * A C function exposed by a `ctypes.CDLL` object
      * An integer specifying the address of a callback function
      * None, in which case any existing callback of that name is removed
  """
  setattr(functions.callbacks, name, new_callback)`

Uptomylimit, 2022-03-17:
Thanks for your anwser!!!!
I can do my own fluid dynamics now!!
At last I use the function in mujoco.wrapper.core like this

mujoco.wrapper.core.set_callback('mjcb_passive',my_callback(env.physics.model,env.physics.data))

Now the robot can move!
And thanks for the answer in this link!
[https://github.com/deepmind/dm_control/issues/7](url)

Uptomylimit, 2022-03-17:
In every step,I set mjpassive callback like this:
`mujoco.wrapper.core.set_callback('mjcb_passive', my_callback(env.physics.model, env.physics.data))`
`def my_callback(mjmodel,mjdata):
    F = np.matrix([[0.0], [0.0], [0.0], [0.0], [0.0], [0.0], [0.0], [0.0], [0.0]])
    F += fluiddynamics_joint9_abs.dragforce_head_joint(env.physics, t)
    F += fluiddynamics_joint9_abs.dragforce_no1_joint(env.physics, t)
    F += fluiddynamics_joint9_abs.dragforce_no2_joint(env.physics, t)
    F += fluiddynamics_joint9_abs.dragforce_no3_joint(env.physics, t)
    F += fluiddynamics_joint9_abs.dragforce_no4_joint(env.physics, t)
    F += fluiddynamics_joint9_abs.dragforce_no5_joint(env.physics, t)
    F += fluiddynamics_joint9_abs.dragforce_tail_joint(env.physics, t)

    mjdata.qfrc_passive[0] += F[0,0]
    mjdata.qfrc_passive[1] += F[1,0]
    mjdata.qfrc_passive[2] += F[2,0]
    mjdata.qfrc_passive[3] += F[3,0]
    mjdata.qfrc_passive[4] += F[4,0]
    mjdata.qfrc_passive[5] += F[5,0]
    mjdata.qfrc_passive[6] += F[6,0]
    mjdata.qfrc_passive[7] += F[7,0]
    mjdata.qfrc_passive[8] += F[8,0]
    print("mjdata",mjdata.qfrc_passive)`

And the result is like this:
`mjdata [ 3.99041754 15.25025445  5.15961426  5.29100649  4.91947889  6.10987817
  8.11888827  7.88103777  3.90188425]`
The second number is the passive force of the hinge joint,nearly 15,but the robot don't move,why?
If i set the qfrc_applied to 15,the robot will move quickly,why?