google-deepmind/mujoco #190: simulate.cc as a pybind11 module

aftersomemath, 2022-03-16:
The release of pybind11 bindings in Mujoco 2.1.2 is very exciting! Would it be reasonable/possible to wrap simulate.cc with pybind11 so that the simulate GUI itself is a Python module?

This would make it very easy for users to work in python and still easily interact with the simulation and change parameters on the fly.

aftersomemath, 2022-03-18:
Thanks for the pointer! I will take look into porting those bindings over. They do not seem too different.

To be clear, my goal is to make a Python driven interactive viewer (click and drag should still work while the simulation is running). It seems this would require two threads to guarantee the UI remains responsive as is done in `simulate.cc` currently.

If that works, then it will be easy to turn the interaction off (passive mode), but then there is thread that doesn't need to be there. With enough refactoring a passive renderer could run on the same thread as the Python process, I don't know if I will have the time to dig into that.

To solve the code duplication problem, the majority of `simulate.cc` could be made into a shared library. However this would mean `simulate.cc` is no longer a self-contained sample (though it is 2k lines already). Does this trade seem reasonable? The only alternative I've thought of so far is a hack requiring `#include "simulate.cc"` 😬.

ciao

yuvaltassa, 2022-03-19:
Yup, Erwin's `mujoco_renderer.cpp` does exactly that, an interactive Python-driven viewer.

We have the same thoughts about breaking out bits of simulate into a library. If refactoring seems like too big a job right now, I reckon that if instead of a PR to this repo, you simply made your viewer available in your own branch, that might be still be useful as a reference to us and to others, as we consider what the best refactor might look like.

closed 2022-03-19 by yuvaltassa