google-deepmind/mujoco #194: How to do visualization using the Python bindings?

BolunDai0216, 2022-03-18:
Hi,

First, I want to say I have little clue of what needs to be done to render the gui. I was simply following the code in the Visualization document, which is

```
// MuJoCo data structures
mjModel* m = NULL;                  // MuJoCo model
mjData* d = NULL;                   // MuJoCo data
mjvCamera cam;                      // abstract camera
mjvOption opt;                      // visualization options
mjvScene scn;                       // abstract scene
mjrContext con;                     // custom GPU context

// ... load model and data

// init GLFW, create window, make OpenGL context current, request v-sync
glfwInit();
GLFWwindow* window = glfwCreateWindow(1200, 900, "Demo", NULL, NULL);
glfwMakeContextCurrent(window);
glfwSwapInterval(1);

// initialize visualization data structures
mjv_defaultCamera(&cam);
mjv_defaultPerturb(&pert);
mjv_defaultOption(&opt);
mjr_defaultContext(&con);

// create scene and context
mjv_makeScene(m, &scn, 1000);
mjr_makeContext(m, &con, mjFONTSCALE_100);

// ... install GLFW keyboard and mouse callbacks

// run main loop, target real-time simulation and 60 fps rendering
while( !glfwWindowShouldClose(window) )
{
    // advance interactive simulation for 1/60 sec
    //  Assuming MuJoCo can simulate faster than real-time, which it usually can,
    //  this loop will finish on time for the next frame to be rendered at 60 fps.
    //  Otherwise add a cpu timer and exit this loop when it is time to render.
    mjtNum simstart = d->time;
    while( d->time - simstart < 1.0/60.0 )
        mj_step(m, d);

    // get framebuffer viewport
    mjrRect viewport = {0, 0, 0, 0};
    glfwGetFramebufferSize(window, &viewport.width, &viewport.height);

    // update scene and render
    mjv_updateScene(m, d, &opt, NULL, &cam, mjCAT_ALL, &scn);
    mjr_render(viewport, &scn, &con);

    // swap OpenGL buffers (blocking call due to v-sync)
    glfwSwapBuffers(window);

    // process pending GUI events, call GLFW callbacks
    glfwPollEvents();
}

// close GLFW, free visualization storage
glfwTerminate();
mjv_freeScene(&scn);
mjr_freeContext(&con);

// ... free MuJoCo model and data
```

I noticed that there is no Python function for `mjr_defaultContext`, `mjv_makeScene` and `mjr_makeContext`, is this a bug or by design? 

If this is by design, how would rendering work in Python? Is it possible to give me some pointers?

Thanks in advance!

rohanpsingh, 2022-03-18:
You can do something like this to create scene and context:
```
scene = mujoco.MjvScene(model, maxgeom=10000)
context = mujoco.MjrContext(model, mujoco.mjtFontScale.mjFONTSCALE_150.value)
```
This will call the default initializer automatically.
You can find more details in the documentation [here](https://mujoco.readthedocs.io/en/latest/python.html).

BolunDai0216, 2022-03-18:
> You can do something like this to create scene and context:
> 
> ```
> scene = mujoco.MjvScene(model, maxgeom=10000)
> context = mujoco.MjrContext(model, mujoco.mjtFontScale.mjFONTSCALE_150.value)
> ```
> 
> This will call the default initializer automatically. You can find more details in the documentation [here](https://mujoco.readthedocs.io/en/latest/python.html). And find an example provided by @saran-t [here](https://github.com/deepmind/mujoco/discussions/185#discussioncomment-2372734).

Thanks for the reply!

I added the lines for creating scene and context. 

However, I still do not see the rendering happening. Here is the code I am using

```
import mujoco as mj

def main():
    max_width = 100
    max_height = 100
    ctx = mj.GLContext(max_width, max_height)
    ctx.make_current()

    cam = mj.MjvCamera()
    opt = mj.MjvOption()

    mj.glfw.glfw.init()
    window = mj.glfw.glfw.create_window(1200, 900, "Demo", None, None)
    mj.glfw.glfw.make_context_current(window)
    mj.glfw.glfw.swap_interval(1)

    mj.mjv_defaultCamera(cam)
    mj.mjv_defaultOption(opt)

    xml_path = "ball.xml"
    model = mj.MjModel.from_xml_path(xml_path)
    data = mj.MjData(model)

    scene = mj.MjvScene(model, maxgeom=10000)
    context = mj.MjrContext(model, mj.mjtFontScale.mjFONTSCALE_150.value)

    while not mj.glfw.glfw.window_should_close(window):
        simstart = data.time

        while (data.time - simstart < 1.0/60.0):
            mj.mj_step(model, data)

        viewport = mj.MjrRect(0, 0, 0, 0)
        mj.glfw.glfw.get_framebuffer_size(window)

        mj.mjv_updateScene(model, data, opt, None, cam, 0, scene)
        mj.mjr_render(viewport, scene, context)

        mj.glfw.glfw.swap_buffers(window)
        mj.glfw.glfw.poll_events()

    mj.glfw.glfw.terminate()

if __name__ == "__main__":
    main()
```

Is it possible to take a look and see what might be the issue?

BolunDai0216, 2022-03-18:
> Try this:
> 
> ```
> import mujoco as mj
> 
> 
> def main():
>     max_width = 100
>     max_height = 100
>     #ctx = mj.GLContext(max_width, max_height)
>     #ctx.make_current()
> 
>     cam = mj.MjvCamera()
>     opt = mj.MjvOption()
> 
>     mj.glfw.glfw.init()
>     window = mj.glfw.glfw.create_window(1200, 900, "Demo", None, None)
>     mj.glfw.glfw.make_context_current(window)
>     mj.glfw.glfw.swap_interval(1)
> 
>     mj.mjv_defaultCamera(cam)
>     mj.mjv_defaultOption(opt)
> 
>     xml_path = "ball.xml"
>     model = mj.MjModel.from_xml_path(xml_path)
>     data = mj.MjData(model)
> 
>     scene = mj.MjvScene(model, maxgeom=10000)
>     context = mj.MjrContext(model, mj.mjtFontScale.mjFONTSCALE_150.value)
> 
>     while not mj.glfw.glfw.window_should_close(window):
>         simstart = data.time
> 
>         while (data.time - simstart < 1.0/60.0):
>             mj.mj_step(model, data)
> 
>         #viewport = mj.MjrRect(0, 0, 0, 0)
>         #mj.glfw.glfw.get_framebuffer_size(window)
>         viewport = mj.MjrRect(0, 0, 1200, 900)
> 
>         #mj.mjv_updateScene(model, data, opt, None, cam, 0, scene)
>         mj.mjv_updateScene(model, data, opt, None, cam, mj.mjtCatBit.mjCAT_ALL.value, scene)
>         mj.mjr_render(viewport, scene, context)
> 
>         mj.glfw.glfw.swap_buffers(window)
>         mj.glfw.glfw.poll_events()
> 
>     mj.glfw.glfw.terminate()
> 
> 
> if __name__ == "__main__":
>     main()
> ```

Thank you so much! It seems like when uncommenting
```
ctx = mj.GLContext(max_width, max_height)
ctx.make_current()
```
the window will not appear.  Would it be possible to give me some pointers to better understand why is that the case?

saran-t, 2022-03-18:
The `mujoco.GLContext` class only provides basic offscreen rendering capability. In your case, since you want to open a GLFW window yourself, you will _not_ want to use `mujoco.GLContext`. Indeed, if you look at https://github.com/deepmind/mujoco/blob/main/python/mujoco/glfw/__init__.py you can see that all it's doing is create an invisible GLFW window to render into, which isn't what you need here.

The snippet suggested by @rohanpsingh looks reasonable, but you should drop the two lines to do with `mujoco.GLContext`.

closed 2022-03-19 by BolunDai0216

rohanpsingh, 2022-03-30:
For future visitors to this thread and until [simulate.cc is wrapped as a Python module](https://github.com/deepmind/mujoco/issues/190), this may be of use to you:  
[mujoco-python-viewer](https://github.com/rohanpsingh/mujoco-python-viewer)

1cookspe, 2022-04-24:
Is it possible to create a Python visualization without using `glfw`? I am trying to embed a Mujoco environment in a PyQt GUI - specifically, in a ```QOpenGLWidget``` widget. From my understanding, `glfw` creates a new window that is separate from my GUI window. Could I instead render the the Mujoco environment in my ```QOpenGLWidget``` widget? Thank you!

JeanElsner, 2022-04-24:
See [here](https://github.com/deepmind/mujoco/discussions/214) for a discussion and running example. The gist of it is that it's certainly possible, however there are some issues with `QOpenGLWidget` in both PySide6 and PyQt6. However if you use `QOpenGLWindow` or `QWindow` it works perfectly.

1cookspe, 2022-04-24:
> See [here](https://github.com/deepmind/mujoco/discussions/214) for a discussion and running example. The gist of it is that it's certainly possible, however there are some issues with `QOpenGLWidget` in both PySide6 and PyQt6. However if you use `QOpenGLWindow` or `QWindow` it works perfectly.

This is perfect! Thank you so much!

yuvaltassa, 2022-11-12:
See #576 for details about the new Python viewer.