google-deepmind/mujoco #195: [Python] Missing arm64 build for macos py3.7

SsnL, 2022-03-18:
currently on M1 it downloads the x86 build which fails obviously

saran-t, 2022-03-18:
Is Python 3.7 actually distributed as prebuilt binaries on M1 machines through any package manager? I've checked python.org and Homebrew and neither provide Python 3.7 as an installable package on M1 machines.

I was under the impression that if you need Python 3.7 on M1 then you're probably building it from source, in which case I expect that you should be able to install `mujoco` via sdist (provided that you have CMake installed).

SsnL, 2022-03-19:
>  Thinking about it a bit more, pip shouldn't be trying to download the x86_64 wheel in the first place if you're running an arm64 build. The fact that it did suggests to me that you might be running an x86_64 build of Python 3.7 via Rosetta2. While the underlying MuJoCo dylib is a universal binary, the x86_64 portion is built with AVX, which can't be run under Rosetta2.

@saran-t Yes you are right about this. Indeed I was using the wrong python build. That's my fault. However, I'd appreciate if it is documented somewhere that Py37 + arm64 is not supported. :)

saran-t, 2022-03-20:
It's not so much that it's not supported as that you'd need to build it from the source distribution. If you have a working version of Python 3.7 pip for arm64, and if you have CMake installed on your system, then `pip install mujoco` should work.

But yes, we can add a section in the README to list the platforms that we provide prebuilt binaries for. Thanks!

closed 2022-03-20 by saran-t