google-deepmind/mujoco #196: Where does the simulator save data and model files?

nikhilweee, 2022-03-19:
I'm running MuJoCo 2.1.0 on MacOS. After loading a model file I tried clicking on "Save Model" or "Save Data" several times but nothing happens. I wonder if this functionality is broken?

yuvaltassa, 2022-03-19:
Hi,

It is not broken. In MacOS those files are saved by default to the `Resources` directory inside the application package (right-click, "Show package contents"). 

This is of course not ideal. What you want is to bring up the system save dialog, which should not be hard to do. If you'd like, you're welcome to send a PR for this.

closed 2022-03-19 by yuvaltassa

nikhilweee, 2022-03-19:
Thanks! I did search in the `bin` directory but didn't think of searching the entire package.