google-deepmind/mujoco #197: How to simulate a substance being broken in two, cut, or smashed?

MotorCityCobra, 2022-03-19:
Does MuJoCo have such bodies?

yuvaltassa, 2022-03-22:
You could create the pieces, weld them together, and disable the welds at run time. This will simulate the effect of breaking.

But more generally no, MuJoCo was not designed for this type of high-dof simulation and is focused rather on articulated systems.

closed 2022-03-22 by yuvaltassa

MotorCityCobra, 2022-04-03:
@yuvaltassa 
Hi, I saw your name on the Maximum a Posteriori Policy Optimisation paper and I wonder if you wouldn't mind me asking what you think is the best algorithm right now for a multi-objective humanoid robot with continuous controls. Sorry, I know this is probably the wrong place to ask. 
I was looking at the [MO-MPO algorithm](http://proceedings.mlr.press/v119/abdolmaleki20a/abdolmaleki20a.pdf).