google-deepmind/mujoco #200: [simulate.cc] Cycling cameras without a loaded model causes a segfault

kevinzakka, 2022-03-23:
On macOS, launching the MuJoCo app and pressing either `[` or `]` without any loaded model causes the window to crash.

<details>
<summary>Toggle traceback</summary>

```bash
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               simulate [52688]
Path:                  /Applications/MuJoCo.app/Contents/MacOS/simulate
Identifier:            org.mujoco.mujoco
Version:               2.1.3 (2.1.3)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-03-23 10:39:15.0658 -0700
OS Version:            macOS 12.1 (21C52)
Report Version:        12
Anonymous UUID:        4D4825B5-3DA0-4BE5-856F-8DDB1A86EF9E

Sleep/Wake UUID:       B9B8A5A1-F823-40B7-BD69-6C3D1ACAAAB5

Time Awake Since Boot: 370000 seconds
Time Since Wake:       1402 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000020
Exception Codes:       0x0000000000000001, 0x0000000000000020
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [52688]

VM Region Info: 0x20 is not in any region.  Bytes before following region: 4298620896
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10037c000-1003ac000    [  192K] r-x/r-x SM=COW  ...acOS/simulate

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   simulate                      	       0x100384a68 uiEvent(mjuiState_*) + 1860
1   simulate                      	       0x1003844a4 uiEvent(mjuiState_*) + 384
2   simulate                      	       0x1003921dc -[GLFWContentView keyDown:] + 232
3   AppKit                        	       0x1b1bc2a78 -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 5920
4   AppKit                        	       0x1b1bc10ec -[NSWindow(NSEventRouting) sendEvent:] + 348
5   AppKit                        	       0x1b1bc0558 -[NSApplication(NSEvent) sendEvent:] + 4064
6   simulate                      	       0x100393540 _glfwPlatformPollEvents + 172
7   simulate                      	       0x100389800 main + 14376
8   dyld                          	       0x100d1d0f4 start + 520

Thread 1:
0   libsystem_pthread.dylib       	       0x1aee2c010 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	       0x1aee2c010 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	       0x1aee2c010 start_wqthread + 0

Thread 4:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	       0x1aedf5954 mach_msg_trap + 8
1   libsystem_kernel.dylib        	       0x1aedf5d00 mach_msg + 76
2   CoreFoundation                	       0x1aeefced8 __CFRunLoopServiceMachPort + 372
3   CoreFoundation                	       0x1aeefb390 __CFRunLoopRun + 1212
4   CoreFoundation                	       0x1aeefa734 CFRunLoopRunSpecific + 600
5   AppKit                        	       0x1b1bbdc90 _NSEventThread + 196
6   libsystem_pthread.dylib       	       0x1aee31240 _pthread_start + 148
7   libsystem_pthread.dylib       	       0x1aee2c024 thread_start + 8

Thread 5:
0   libsystem_kernel.dylib        	       0x1aedf85c0 __psynch_mutexwait + 8
1   libsystem_pthread.dylib       	       0x1aee2e364 _pthread_mutex_firstfit_lock_wait + 84
2   libsystem_pthread.dylib       	       0x1aee2bc98 _pthread_mutex_firstfit_lock_slow + 240
3   libc++.1.dylib                	       0x1aed81ca4 std::__1::mutex::lock() + 16
4   simulate                      	       0x100385cfc simulate() + 280
5   simulate                      	       0x100389efc void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, void (*)()> >(void*) + 44
6   libsystem_pthread.dylib       	       0x1aee31240 _pthread_start + 148
7   libsystem_pthread.dylib       	       0x1aee2c024 thread_start + 8

Thread 6:
0   libsystem_pthread.dylib       	       0x1aee2c010 start_wqthread + 0

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000000   x1: 0x00000000ffffffff   x2: 0x00000000ffffffff   x3: 0x0000000000000000
    x4: 0x000000016fa7d76c   x5: 0x0000000000000035   x6: 0x000060000042ea00   x7: 0x0000000000000000
    x8: 0x0000000000000000   x9: 0xec77eb29a21a0015  x10: 0x0000000000000004  x11: 0x000000010148714c
   x12: 0x00000001013b4c84  x13: 0x000000000000005a  x14: 0x0000000000000000  x15: 0x00000001009dec30
   x16: 0x00000001013b49b0  x17: 0x00000000000001ee  x18: 0x000000012030b940  x19: 0x00000001009dec10
   x20: 0x0000600001f1b000  x21: 0x00000001007988c8  x22: 0x000000012a907210  x23: 0x00000001f940261a
   x24: 0x0000000000000100  x25: 0x000000000000000a  x26: 0x00000001f93dafbd  x27: 0x00000000000000fc
   x28: 0x000000002bd80025   fp: 0x000000016fa7ee20   lr: 0x00000001003844a4
    sp: 0x000000016fa7d900   pc: 0x0000000100384a68 cpsr: 0x60001000
   far: 0x0000000000000020  esr: 0x92000006 (Data Abort) byte read Translation fault
```

</details>

yuvaltassa, 2022-03-23:
Yup. 
Thanks @kevinzakka for catching this!
This is now fixed internally. 
Until we push out a fix, please refrain from hitting those keys before loading a model :)

yuvaltassa, 2022-03-23:
Actually sorry for closing.
I should leave this open until fix in the public repo.

saran-t, 2022-03-23:
Fixed in 21338e61784eebea1ae850a1cdb3bcf049de5924. Released binary packages have not yet been updated to reflect this change though.

saran-t, 2022-03-30:
I'm going to close the issue this is fixed in source. Binary will be updated at our next release point, but please do let us know if there's a strong case for us to do so sooner.

closed 2022-03-30 by saran-t