google-deepmind/mujoco #202: How to access mjuiState in python?

rohanpsingh, 2022-03-24:
I've been trying to build a simple renderer for mujoco python [here](https://github.com/rohanpsingh/mujoco-python-viewer).  
It's basically the same as the mujoco-py viewer but now I would like to make it interactive like simulate.cc

For this it would be helpful if I could access mjuiState.

rohanpsingh, 2022-03-30:
I'm still struggling with this, so if somebody can comment on a solution, I'd be really grateful.

JeanElsner, 2022-03-30:
I think the ui functions and enums are not part of the bindings, as far as I can tell from the code generation. However you don't really need them anyway, there are many good ui frameworks for python (I don't think the mujoco-py viewer uses any of the ui API anyway, rather it uses glfw for user interaction). You might be interested in [this](https://github.com/deepmind/mujoco/pull/201) (someone planning to do python bindings for simulate.cc) or [this](https://github.com/deepmind/mujoco/discussions/214) (embedding renderer in ui framework).

rohanpsingh, 2022-04-04:
> I think the ui functions and enums are not part of the bindings

Thanks for confirming. I guess I shouldv'e looked into the code generation myself before opening the issue. 
I was already using glfw in my project but thought having access to mjui will make some user interactions easier to handle.

Nevertheless, I ended up implemeting the mouse interactions (to apply perturbations) using glfw only here: https://github.com/rohanpsingh/mujoco-python-viewer 
It does the job for me, but some users might still be looking to have python bindings for simulate.cc

> But since you've requested, I'll put mjui Python bindings on our TODO list.

Thanks a lot @saran-t , but I don't need that anymore so perhaps you can strike it off your TODO list :)

closed 2022-04-04 by rohanpsingh