google-deepmind/mujoco #208: How to build from source on M1 Mac?

BolunDai0216, 2022-03-25:
Hi,

I was wondering if there are any instructions for building from source on a M1 Mac? If there is can someone kindly point me to it?

Thanks in advance!

BolunDai0216, 2022-03-25:
> For the Python bindings, the build instructions [in our docs](https://mujoco.readthedocs.io/en/latest/python.html#building-from-source) apply to M1 Macs.
> 
> For the sample applications, the Makefiles provided work on all Macs.

Thanks for the quick reply!

After downloading the source code for 2.1.3 I executed `make` in the sample file and I got the following error

```
c++ -O2 -I../include -L../lib -std=c++11 -pthread -Wl,-rpath,'$ORIGIN'/../lib testxml.cc    -lmujoco_nogl               -o ../bin/testxml
ld: warning: directory not found for option '-L../lib'
ld: library not found for -lmujoco_nogl
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [all] Error 1
```

Is it possible to help me with this?

saran-t, 2022-03-25:
What directory are you running `make` from? The Makefile assumes that you're running from the `sample` directory in the mounted DMG, or a directory containing all files from the DMG.

BolunDai0216, 2022-03-25:
> What directory are you running `make` from? The Makefile assumes that you're running from the `sample` directory in the mounted DMG, or a directory containing all files from the DMG.

I'm running it under the sample directory. What does a mounted DMG mean?

I did not download the .dmg file. I was using the .zip file.

saran-t, 2022-03-25:
The directory structure you get if you open up the DMG file is slightly different from what you get from the zip. The MuJoCo binary itself isn't in the zip file (it's auto-generated by GitHub from the repo contents) so you won't be able to build without downloading the DMG to get the precompiled library anyway.

If you want to build from the zip archive, you should be able to do so by placing `MuJoCo.framework` from the DMG one directory up from `sample`.

BolunDai0216, 2022-03-25:
> The directory structure you get if you open up the DMG file is slightly different from what you get from the zip. The MuJoCo binary itself isn't in the zip file (it's auto-generated by GitHub from [...]

I see. Thank you so much for explaining this!

BolunDai0216, 2022-03-25:
I was able to run `make` and I can see all the binaries generated.

However, when I tried to run `./simulate` in the sample folder I get

```
dyld[10276]: Library not loaded: @rpath/MuJoCo.framework/Versions/A/libmujoco.2.1.3.dylib
  Referenced from: /Users/bolundai/.mujoco/mujoco-2.1.3/sample/simulate
  Reason: tried: '/usr/local/lib/libmujoco.2.1.3.dylib' (no such file), '/usr/lib/libmujoco.2.1.3.dylib' (no such file)
[1]    10276 abort      ./simulate
``` 

Do you know what this issue is?

kevinzakka, 2022-03-27:
@saran-t I can run `make all` on my M1 but not execute the simulate binary with the same error as @BolunDai0216. Only after copying `libmujoco.2.1.3.dylib` into `/usr/local/lib` does it work.

saran-t, 2022-03-31:
Can you please try adding the following to `ALLFLAGS` in the macOS Makefile?

`-Wl,-rpath,@executable_path/..`

closed 2022-04-08 by saran-t