google-deepmind/mujoco #215: Looking for a simple example for python binding

mosem, 2022-03-30:
I'm trying to run a simple script to get started, and it would help if there was one available. Any chance to add an example script?

At the moment I'm running this and all I'm getting is a black screen (maybe my camera is looking at the wrong place, or my model is empty?). Any help would be appreciated.

```
import mujoco
import glfw

def init_window(max_width, max_height):
    glfw.init()
    # glfw.window_hint(glfw.VISIBLE, 0)
    window = glfw.create_window(width=max_width, height=max_height,
                                       title='Demo', monitor=None,
                                       share=None)
    glfw.make_context_current(window)
    return window

window = init_window(1200, 900)

model = mujoco.MjModel.from_xml_path('./xml_files/ball.xml')
data = mujoco.MjData(model)
context = mujoco.MjrContext(model, mujoco.mjtFontScale.mjFONTSCALE_100)
mujoco.mjr_setBuffer(mujoco.mjtFramebuffer.mjFB_OFFSCREEN, context)

scene = mujoco.MjvScene(model, 1000)
camera = mujoco.MjvCamera()

mujoco.mjv_updateScene(
    model, data, mujoco.MjvOption(), mujoco.MjvPerturb(),
    camera, mujoco.mjtCatBit.mjCAT_ALL, scene)

while(not glfw.window_should_close(window)):
    while data.time < 1:
        mujoco.mj_step(model, data)
        print(data.geom_xpos)

    glfw.get_framebuffer_size(window)

    viewport = mujoco.MjrRect(0, 0, 0, 0)

    mujoco.mjv_updateScene(
        model, data, mujoco.MjvOption(), None,
        camera, mujoco.mjtCatBit.mjCAT_ALL, scene)
    mujoco.mjr_render(viewport, scene, context)

    glfw.swap_buffers(window)

    glfw.poll_events()

glfw.terminate()
```

Together with this xml file:
```
<mujoco>
    <worldbody>
            <light diffuse=".5 .5 .5" pos="0 0 3" dir ="0 0 -1"/>
        <geom type="plane" size="1 1 0.1" rgba=".9 0 0 1" />
        <body pos="0 0 1">
            <joint type="free"/>
            <geom type="sphere" size=".1" rgba="0 .9 0 1"/>
        </body>
    </worldbody>
</mujoco>
```

saran-t, 2022-03-30:
I was going to post exactly what @JeanElsner, but then I tried it on my laptop (macOS) and it didn't work for me.

@mosem  Please go ahead and try it and let us know if it works for you.

kevinzakka, 2022-03-30:
@saran-t Wasn't working for me either on macOS but that's because @mosem reinits the viewport inside the while loop. Moving it to before the while loop works.

kevinzakka, 2022-03-30:
Working code:

```python
import mujoco
import glfw

def init_window(max_width, max_height):
    glfw.init()
    window = glfw.create_window(width=max_width, height=max_height,
                                       title='Demo', monitor=None,
                                       share=None)
    glfw.make_context_current(window)
    return window

window = init_window(1200, 900)

model = mujoco.MjModel.from_xml_path('model/mug/mug.xml')
data = mujoco.MjData(model)
mujoco.mj_forward(model, data)
context = mujoco.MjrContext(model, mujoco.mjtFontScale.mjFONTSCALE_100)

width, height = glfw.get_framebuffer_size(window)
viewport = mujoco.MjrRect(0, 0, width, height)

scene = mujoco.MjvScene(model, 1000)
camera = mujoco.MjvCamera()
mujoco.mjv_updateScene(
    model, data, mujoco.MjvOption(), mujoco.MjvPerturb(),
    camera, mujoco.mjtCatBit.mjCAT_ALL, scene)

while(not glfw.window_should_close(window)):
    while data.time < 1:
        mujoco.mj_step(model, data)
        print(data.geom_xpos)

    glfw.get_framebuffer_size(window)

    mujoco.mjv_updateScene(
        model, data, mujoco.MjvOption(), None,
        camera, mujoco.mjtCatBit.mjCAT_ALL, scene)
    mujoco.mjr_render(viewport, scene, context)

    glfw.swap_buffers(window)
    glfw.poll_events()

glfw.terminate()
```

JeanElsner, 2022-03-30:
Thanks @kevinzakka that's good to know. I updated my own example accordingly.

saran-t, 2022-03-30:
@kevinzakka in your first comment did you literally just moved `viewport = mujoco.MjrRect(0, 0, width, height)` outside of the loop? If that changed anything then it's super weird since the viewport struct shouldn't be saved into any other struct.

kevinzakka, 2022-03-30:
@saran-t you're right, I got confused by 2 simultaneous changes. Moving the viewport doesn't make a different.

What does make the difference is assigning the height, width variables and commenting out the offscreen buffer!

saran-t, 2022-03-30:
OK, that makes more sense. Those are the things I did on my side too but I just ended up with a blinking red window. Didn't spend very long on it though, and it sounds like you also have it working as well.

saran-t, 2022-03-31:
@mosem did @kevinzakka's example work for you?

mosem, 2022-03-31:
Yes!
Thank you very much for the help and support.

closed 2022-03-31 by mosem

dhruvthanki, 2022-04-09:
I tried running the same code (@kevinzakka) but doesn't seem to work on my end. I get the following error:
```
Traceback (most recent call last):
  File "/home/dhruv/Documents/Digit-py/src/main.py", line 16, in <module>
    context = mujoco.MjrContext(model, mujoco.mjtFontScale.mjFONTSCALE_100)
mujoco.FatalError: gladLoadGL error
```
Cant seem to figure out whats wrong.

Barbany, 2022-04-14:
@dhruvthanki it seems it's an ongoing issue of the latest MuJoCo release (https://github.com/deepmind/mujoco/issues/217). I had the same problem and solved it using the previous version, i.e., `pip install mujoco==2.1.3 `.

This solves the issue with GLAD. You can test it by running `python -m mujoco.render_test`.

saran-t, 2022-04-14:
@Barbany @dhruvthanki The 2.1.5 release from yesterday should've resolved this.

Barbany, 2022-04-20:
@saran-t the 2.1.5 release indeed resolved this issue for my desktop computer.

However, I was trying to setup MuJoCo in a headless server and when I run the `render_test`I get the previously reported `mujoco.FatalError: GLEW initialization error: Missing GL version` error.

saran-t, 2022-04-20:
If you're getting an error that says GLEW it means that you're running `mujoco<=2.1.3`.

Barbany, 2022-04-20:
You're right I copied the wrong error code. With `mujoco==2.1.5` on my server while connected with X forwarding I get:
```
Running tests under Python 3.9.12: /home/obarbany/miniconda3/envs/mujoco/bin/python3
[ RUN      ] MuJoCoRenderTest.test_can_render
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
/home/obarbany/miniconda3/envs/mujoco/lib/python3.9/site-packages/glfw/__init__.py:906: GLFWError: (65543) b'GLX: Failed to create context: BadValue (integer parameter out of range for operation)'
  warnings.warn(message, GLFWError)
[  FAILED  ] MuJoCoRenderTest.test_can_render
[ RUN      ] MuJoCoRenderTest.test_safe_to_free_context_twice
[  FAILED  ] MuJoCoRenderTest.test_safe_to_free_context_twice
======================================================================
ERROR: test_can_render (__main__.MuJoCoRenderTest)
MuJoCoRenderTest.test_can_render
Test that the bindings can successfully render a simple image.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/obarbany/miniconda3/envs/mujoco/lib/python3.9/site-packages/mujoco/render_test.py", line 54, in test_can_render
    context = mujoco.MjrContext(self.model, mujoco.mjtFontScale.mjFONTSCALE_150)
mujoco.FatalError: gladLoadGL error

======================================================================
ERROR: test_safe_to_free_context_twice (__main__.MuJoCoRenderTest)
MuJoCoRenderTest.test_safe_to_free_context_twice
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/obarbany/miniconda3/envs/mujoco/lib/python3.9/site-packages/mujoco/render_test.py", line 87, in test_safe_to_free_context_twice
    context = mujoco.MjrContext(self.model, mujoco.mjtFontScale.mjFONTSCALE_150)
mujoco.FatalError: gladLoadGL error

----------------------------------------------------------------------
Ran 2 tests in 0.954s
```

If not using X forwarding, I get:
```
Running tests under Python 3.9.12: /home/obarbany/miniconda3/envs/mujoco/bin/python3
[ RUN      ] MuJoCoRenderTest.test_can_render
/home/obarbany/miniconda3/envs/mujoco/lib/python3.9/site-packages/glfw/__init__.py:906: GLFWError: (65544) b'X11: The DISPLAY environment variable is missing'
  warnings.warn(message, GLFWError)
/home/obarbany/miniconda3/envs/mujoco/lib/python3.9/site-packages/glfw/__init__.py:906: GLFWError: (65537) b'The GLFW library is not initialized'
  warnings.warn(message, GLFWError)
python3: /builds/florianrhiem/pyGLFW/glfw-3.3.7/src/posix_thread.c:64: _glfwPlatformGetTls: Assertion `tls->posix.allocated == 1' failed.
Fatal Python error: Aborted
```

saran-t, 2022-04-20:
If you want headless GPU rendering you should set the environment variable `MUJOCO_GL=egl`.

Barbany, 2022-04-20:
I get the exact same error before and after setting the environment variable `MUJOCO_GL=egl` with and without X forwarding. Should I open a new issue/share some info about my machine/...?

saran-t, 2022-04-20:
@Barbany Yes, please open a new issue about that with a bit more info. Thanks!