google-deepmind/mujoco #216: Calculating optimal muscle length

studywolf, 2022-03-30:
Hello! 

I'm working on a project that involves modeling a muscle based system, and we need to know the optimal muscle length that's used by Mujoco. In the [documentation](https://mujoco.readthedocs.io/en/latest/modeling.html#cmuscle), it says

"We assume that the biological tendon is inelastic, with constant length LT...

The length range computation described above already provided the operating range for LT + LM. In addition, we ask the user to specify the operating range for the muscle length LM scaled by the (still unknown) constant L0. This is done with the attribute range; the default scaled range is (0.75, 1.05). Now we can compute the two constants, using the fact that the actual and scaled ranges have to map to each other:

(actuator_lengthrange[0] - LT) / L0 = range[0]
(actuator_lengthrange[1] - LT) / L0 = range[1]"

So the task is tracking down which parameters are the proper ones to use in this calculation. So, after rearranging the above equations to solve for `LT` and then substitute and solve for `L0`, and scouring the API, the following is what i'm using: 
```
lengthrange = sim.model.actuator_lengthrange
range[0] = sim.model.actuator_gainprm[0]
range[0] = sim.model.actuator_gainprm[0]

L0 = lengthrange[:,0] / (range[:,0] + range[:,1] + lengthrange[:,1])
```

The thing that is confusing is that when we try to calculate `L0` using the original equations, we get different answers. This could be because we're using the wrong value for `LT`, or we're using the wrong values for one of the variables somewhere else. 
```
        LT = sim.model.tendon_length0  # tendon length in qpos0
        # the following equations give different answers for L0, but should be the same
        L0 = (lengthrange[:,0] - LT) / gainprm[:,0]
        L0 = (lengthrange[:,1] - LT) / gainprm[:,1]
```

So I'm hoping that someone can clarify what the appropriate parameters are to use in this calculation! `LT` in particular I'm unsure of. Or, even better, if they could point me towards where in the simulation the actual `L0` value that is calculated and used is stored.
Thanks!

yeshaokai, 2022-03-30:
Same question here! Would be great if someone can help!!!

vittorione94, 2022-04-02:
Hello,

Let me first answer why the two computations of L0 don't match. In MuJoCo tendons are a very general  thing, you can think of them as strings connecting two points and they're **not muscles nor biological tendons**. They become muscles when we attach an actuator of type muscle on that tendon. 

So the value stored in tendon_length0 is exactly the same as actuator_length at time 0 (in the initial qpos) and **not LT**. The values of actuator_length should be always within the actuator_lengthrange (now I say should because you could set the model in a weird joint position and the value might be outside).
For reference but not super relevant for this discussion, there's a way to make tendon_length0 and actuator_length different and that's by using **gear** in the muscle, gear is a way of scaling the muscle length for example a gear=5 will make the actuator_length 5 times larger than tendon_length0.

You can compute L0 and LT in a similar way to what you tried before but as far as I know MuJoCo doesn't expose the computed values of L0 and LT to check that what we have is correct (This is probably the biggest difference between MuJoCo and OpenSim). If they were exposed they should be in the model since they don't change through out the simulation.

The way I computed L0 is `(lengthrange[:,1] - lengthrange[:,0]) / (range[:,1] - range[:,0]) ` 

My reasoning is the following:
`(LR_0 - LT) / L0 = R0`
`(LR_0 - LT) = R0*L0`
`LT = -(R0*L0 - LR_0)`
`LT = LR_0 - R0*L0`

you can substitute in the other eqn.
`(LR_1 - (LR_0 - R0*L0)) / L0 = R1`
`LR_1 - (LR_0 - R0*L0)  = R1*L0`
`LR_1 = R1*L0 + (LR_0 - R0*L0)`
`LR_1 = (R1 - R0)*L0 + LR0`
`LR_1 - LR0 = (R1 - R0)*L0`
`(LR_1 - LR0) / (R1 - R0) = L0`

I hope this makes sense and helps the discussion :) 

Cheers!

studywolf, 2022-04-08:
Hello, thank you so much for your response! I wasn't hopeful (had this question open on the Mujoco forums for a while).

So, we actually have been computing L0 this way, found exactly the way you described! We ended up questioning this because it lead to some strange behavior in our muscle spindle model. I'll actually copy/paste the forum question as it provides the whole context for my current question (updated with my current understanding of how to calculate each parameter:

> Hello,
I am working to implement the muscle spindle model described in [this paper](https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=9089841). The parameters that it needs are
>
> **FM** : The force exerted by the muscle fiber FM (N)
**FMmax** : Maximal isometric force (N)
**lM** : Fiber length (mm)
**lMopt** : Optimal fiber length (mm) == rest sarcomere length
**a** : Normalized muscle activity
**vM** : Fiber contraction velocity (mm/s)
>
> I'm working on getting these values out of Mujoco. So far I'm looking at
>
>  **FM** : The force exerted by the muscle fiber FM (N)
`-1 * self.interface.sim.data.actuator_force`
>
> **FMmax** : Maximal isometric force (N)
the user specified 'force' if specified (`gainprm[ii][2]`), otherwise `gainprm[ii][3] / interface.sim.model.actuator_acc0[ii])`
>
> **lM** : Fiber length (mm)
want to calculate the actual muscle length, my understanding is actuator length is muscle length + tendon length, so take the actuator length, subtract out the resting state tendon length, add in `gainprm[:, 5]`, which is the minimum length of the muscle (`gainprm[:, 6]` is the max)
 `(self.interface.sim.data.actuator_length - self.interface.sim.model.tendon_length0 + self.interface.sim.model.actuator_gainprm[:, 5])`
>
> **a** : Normalized muscle activity
`self.interface.sim.data.act` ... ? 
>
> **vM** : Fiber contraction velocity (mm/s)
The fiber contraction velocity I'm guessing in this case is `sim.data.actuator_velocity` ... ? There's also `ten_velocity`, and it's not clear to me how they interact, exactly.

So, I can scratch the lMopt calculation off the list, are you able to shed light onto getting the other parameters as well?? 
Any insight that you could provide would be greatly appreciated!
Thanks,

vittorione94, 2022-04-12:
Hello again 😃

I'm happy that our calculations lead to the same result 🥳
So going back to the parameters:

**FM** and **FMmax** I'd say are what you say they are. 

The **Fiber length** (assuming we want the muscle length that should change through out the simulation since the muscle stretches) is different from what I would compute. After computing L0 we can then compute LT as: 
`LT = LR_0 - R0*L0` (and this should be constant because the biological tendon is assumed to be inelastic). Then at runtime I would take  actuator_length and subtract LT from it to to get the fiber length. **Note**: you might get negative values for your biological tendon since L0 and LT are found via optimization in MuJoCo, there was a question in the previous forum about this ([negative biological tendon length](https://roboti.us/forum/index.php?threads/can-lt-length-of-tendon-have-a-negative-value.4143/)) if that happens what I would do is to play with the length_ranges or ranges or the muscle routing which you can tune to get values that you are happy with. 

**Normalized muscle activity** so just to have the same nomenclature in biomechanics we have excitations and activations. Now, let's say you're controlling a muscle in MuJoCo what you're actually feeding in the simulation is an  excitation, that value is then converted in an activation using the muscle dynamics (I'm not referring to how the FL and FV curves are implemented but the tau function in the documentation, which specifies how the activation of a muscle changes through time) in MuJoCo. And yes as you said, the muscle activation is stored in data.act.

**Fiber contraction velocity** The measurement unit varies on your simulation params since it definitely depends on what do you pass in the lengthrange as this value is dependent on it and the time your simulation is currently running at. The contraction velocity is stored in either actuator_velocity or ten_velocity, here same principle applies to what I explained in my previous answer on the relationship between actuator_length and ten_length, so they are exactly the same unless you start scaling the actuator length using gears (which in my experience with muscle simulation should never do because it causes all sort of inconsistencies).

Hope this clarifies some doubts!
Cheers!

studywolf, 2022-05-03:
This is excellent, thank you very much. Have been trying to sort this out for a long time!

studywolf, 2022-05-03:
Actually, quick follow up question: for the fibre length, I think what I actually want is the scaled muscle length `L`, since that's the value that's passed into the force generating function that Mujoco uses...is that correct? That the force is calculated as a function of `L` and not `LM`?

studywolf, 2022-06-14:
Reopening this just as a bump for my last question!

vittorione94, 2022-06-14:
I unassigned myself by mistake lol

studywolf, 2022-06-15:
Hello! This is great. I keep forgetting that it's open source now and I can actually just go and look at the code. Super helpful; I am convinced all my questions are answered. Thanks again!

closed 2022-06-15 by studywolf