google-deepmind/mujoco #218: modelling issue of ball in cup

xibeisiber, 2022-04-01:
Hi all,
I'm simulating the ball in cup problem (picture 1) and encounter some problems in modelling.

If the arm, cup, rope are all in one body connected with joints, they can behave normally if I send torque command to the joints of the arm. The problem is that the command trajectory can't be followed up to expectation when using the PD law plus gravity compensation (adjust PD gain may help). When I try to use `set_joint_qpos` to send position commands to the arm joints directly, the rope is floating (see picture 2)...

If the arm and cup are one body, a mocap is attached to the cup bottom, then one end of the rope is fixed on the mocap using `equality/weld `, the `set_joint_qpos` method works,  but when the cup tilts, the rope would pass through the cup(see picture 3)..

So my question is if I want to use position control of the arm joints by `set_joint_qpos`, how should I modify the model?

Thanks for any help!
[image]
[image]
[image]

nimrod-gileadi, 2022-04-04:
I'm not too familiar with the mujoco-py functions, but setting joint qpos is not equivalent to position control, since it skips physics simulation: It's not asking the the arm to go to a position - it's telling the arm that it's already at that position, which might be inconsistent with constraints.

I think that the direction you suggested of adjusting the PD gain sounds reasonable, or you could try replacing the `<motor>` actuators in your model with appropriate `<position>` actuators and set control inputs instead of qpos directly.

xibeisiber, 2022-04-05:
@nimrod-gileadi 
Thanks very much for your reply. I'm a bit confused about what`actuator/position` does. Is it actually a P controller? I saw somewhere `actuator/position` should be used together with `actuator/velocity` so that it is a PD controller..

By using the PD law (C(q,qdot)+M(acc+kp*q_err+kd*qdot_err)), I found previously the reason that it's hard to tune the kp and kd may be the my XML file for the manipulator doesn't have accurate mass and inertia information. I noticed that the mujoco document mentions the "system identification". What does it do? Can the "system identification" calculate more accurate mass and inertia information?

xibeisiber, 2022-04-07:
@JeanElsner ok. thanks a lot~

closed 2022-04-07 by xibeisiber