google-deepmind/mujoco #219: Help with mujoco-py installation

JodieKoenig, 2022-04-01:
When I try : pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
The following error is coming up...

Using legacy 'setup.py install' for mujoco-py, since package 'wheel' is not installed.
Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... error
  error: subprocess-exited-with-error

Later followed by legacy-install-failure

Any suggestions or help would be greatly appreciated
Thanks, Jodie

JodieKoenig, 2022-04-01:
Thank you for your prompt response. Yes, I am trying to use MuJoCo via Python to train an agent on Open AI gym environments. I will have a look at the official Python bindings

saran-t, 2022-04-04:
FYI: there seems to be an active ongoing effort to migrate `gym` away from `mujoco-py` and to our new `mujoco` bindings package instead. https://github.com/openai/gym/pull/2595

saran-t, 2022-04-09:
I'm going to close this issue for now, but looking back at your original message I _think_ you just need to `pip install wheel` to make progress. If you want to go via that route then you should follow the instructions given in the `gym` and `mujoco-py` repos to install an older version of MuJoCo.

closed 2022-04-09 by saran-t