google-deepmind/mujoco #222: How mujoco handles inverse kinematics issues?

mch5048, 2022-04-05:
Hi, thanks for sharing this API public to the world!

Currently, I am working on deep robotics RL project using CoppeliaSim (V-REP).
CoppeliaSim has its own inverse kinematics solver and it seems the solver struggles with 
noisy target poses generated by the RL policy if I set the action space of the agent as Cartesian pose.
I found many people also found issues with this IK problems when they are doing RL in the continuous action space.

However, in the papers or the projects that uses MuJoCo, it seems no one finds issues with the RL policy making infeasible 
motions in the Cartesian space during the exploration.

So, I wonder if there are special features that handles when solving IK. Does it make infeasible action in the real world possible in the simulated environment? 

Thanks in advance!

JeanElsner, 2022-04-05:
Actually MuJoCo as a simulator doesn't do IK solving for you. Rather you provide a control signal for your robot's actuators. It's then up to you to implemend e.g. an operational space controller or some form of motion generator. MuJoCo however can be quite useful to do differential computations. I found this bit of code from dm_control quite handy to get started using MuJoCo to do IK:

https://github.com/deepmind/dm_control/blob/master/dm_control/utils/inverse_kinematics.py

Balint-H, 2022-04-06:
Although it is not applicable for many cases, for some applications that IK may be needed you could also weld end effectors to mocap bodies, or use other equality constraints to implicitly define the kinematic chain with relation to a reference. This may be suitable if you just want to generate reference kinematic poses for a dynamic agent to follow.

JeanElsner, 2022-04-06:
Having a quick look at meta world I can see that they did exactly what @Balint-H proposed for the implementation of the Sawyer model (cf. [here](https://github.com/rlworkgroup/metaworld/blob/master/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py)). d4rl seems to be based on OpenAI Gym and taking a look at the [MuJoCo environment in Gym](https://github.com/openai/gym/blob/07fd227a4fc0ecbd4fdac5563c0f48deeb5688ed/gym/envs/mujoco/mujoco_env.py#L44) tells me that the action space is simply the MuJoCo actuators, so no IK at all.

mch5048, 2022-04-07:
@Balint-H @JeanElsner  mocap it is! I wonder if the policy learned with mocap-based Cartesian motion can be generalized to the real world robot that should solve IK. Thanks for the response!

closed 2022-04-07 by mch5048