google-deepmind/mujoco #227: Joint snaps to lower range once simulation starts?

livinamuk, 2022-04-07:
I have a simple hinge joint defined with

`<joint type="hinge" axis="0 1 0" range="-59 59" damping="0" stiffness="0" armature=".01" limited="true" solimplimit="01 .99 .01"/>`

and I can't understand why it _snaps_ to its lower range once the simulation starts. Setting the range to `"0 59"` stops the snapping obviously but I need to be able to set the desired range. What is it about setting the minimum here that forces the thigh to the position shown? (left leg removed for simplicity)

I've tried playing with the stiffness, damping, and solimplimit attributes and have come to undertsand it's something I don't.

[image]

video https://youtu.be/wiZ_B39Vi5c

xml https://www.principiaprogrammatica.com/dump/test.xml

yuvaltassa, 2022-04-07:
Looks like a self-collision? Hit the 'C' and/or 'F' buttons in `simulate` to verify.

Also read [this](https://github.com/deepmind/mujoco/issues/104#issuecomment-1007323284) for a more thorough explanation. Perhaps this text should be upgraded to the documentation.

closed 2022-04-07 by yuvaltassa

livinamuk, 2022-04-07:
That was it, thank you. Self collision is enabled when the root is static.

livinamuk, 2022-04-25:
Is it possible to disable self collision in this case?

yuvaltassa, 2022-04-25:
Either that or use the [exclude](https://mujoco.readthedocs.io/en/latest/XMLreference.html#contact-exclude) clause, which might be more convenient.