google-deepmind/mujoco #229: Try to simulate a suction-type gripper

Lucky-hqj, 2022-04-08:
I want to try to simulate a suction-type gripper. When the suction-type gripper is close enough to the object, the object can be sucked. I would like to ask what characteristics I can use to achieve these functions？Thanks for any help!

yuvaltassa, 2022-04-08:
We are considering adding this type of actuator which would work via contact points. So _after_ contact has been achieved the actuator will be able to _inject an adhesive force via the contact normal_. Does that sound like it would do what you want, or do you require attractive forces at a distance (i.e. before contact is made)?

If the latter, there another way to do this, by activating soft weld constraints when the body is close enough, but this would be more of an on/off actuator, rather than a variable force, as you would have in the first proposal.

Is anyone else interested in this type of actuation? Please reply to this thread if yes.

Lucky-hqj, 2022-04-09:
> We are considering adding this type of actuator which would work via contact points. So _after_ contact has been achieved the actuator will be able to _inject an adhesive force via the contact [...]

Thanks a lot for your answer. As you said, the first suggestion was exactly what I was looking for. If this proposal is implemented, a lot of interesting things can be done. Until then, I should probably take your second suggestion and try to temporarily implement my functionality with soft weld constraints. But I'm very much looking forward to what you're saying about the adhesive force. Thanks again for your contribution to this work. @yuvaltassa

renatyv, 2022-07-05:
Would love to see this proposal implemented. Is there any progress on this feature?

yuvaltassa, 2022-07-06:
No progress yet, but thanks for your interest, we will bump this up on our priority list.

Would you mind saying a few words about your use case?

renatyv, 2022-07-06:
We are trying to implement DL for a robotic arm with a suction gripper at the end. Currently looking at the weld constraint, but specialized solution would be nicer

yuvaltassa, 2022-07-06:
What is DL?

hendrikE, 2022-07-06:
I'm looking for a similar function as well! My usecase is different though, I'm working with climbing robots and would like to subtitute the actual grippers with "dummy" grippers in the simulation to decrease the computational load.

yuvaltassa, 2022-07-06:
Hi @hendrikE,

That is a somewhat easier request. In order to do that you need to flip `eq_active` to turn on your connect equality constraint (I assume you want a connect? a weld is also possible but we need to fix some things about welds and until then I'm not sure it will work as well). Below is a utility function that I believe should do the correct thing, though I have not tested it. Given an already defined connect constraint between two bodies, it will move the anchor point to a global point `pos` such that the constraint will be exactly satisfied given the current state of `mjData`. This should work well if the two body IDs are fixed e.g. a hand and the world. That should let you do something similar to the Naderi et al. work ([paper](https://onlinelibrary.wiley.com/doi/abs/10.1111/cgf.13513), [paper](https://dl.acm.org/doi/10.1145/3072959.3073707), [video](https://www.youtube.com/watch?v=HD698J2ww4w)), is this what you are thinking of?

```C
// set anchor of connect constraint i to global position pos
void mj_setConnectAnchor(mjModel* m, const mjData* d, int i, mjtNum pos[3]) {
  if (m->eq_type[i]!=mjEQ_CONNECT) {
    mju_error("equality must be a 'connect' for mj_setConnectAnchor");
  }

  // data[0-2] = anchor position in body1 local frame
  int id1 = m->eq_obj1id[i];
  mju_subFrom3(pos, d->xpos+3*id1);
  mju_rotVecMatT(m->eq_data+mjNEQDATA*i, pos, d->xmat+9*id1);

  // data[3-5] = anchor position in body2 local frame
  int id2 = m->eq_obj2id[i];
  mju_subFrom3(pos, d->xpos+3*id2);
  mju_rotVecMatT(m->eq_data+mjNEQDATA*i+3, pos, d->xmat+9*id2);
}
```

Note this function is completely untested, let me know if you find that it works (or doesn't 🙂 )

hendrikE, 2022-07-07:
Hi @yuvaltassa,

That's brilliant, thank you so much! I'll try it out and let you know if it works! I'm not entirely sure what the difference between the weld and the connection would be (would the weld be soft?), but I think this should definitely work.

yinzikang, 2022-07-19:
Hi, I am doing some similar work these days. What I am working at is making a gripper move down, connect a box with the "connect" equality, and then move up. 

So the question is: how can I calculate the force which should equal the box's gravity between them? 

I have tried using the force sensor in several ways but none of them worked. I thought it was because that the gripper and the box do not contact at all due to the spring characteristic of the "connect" equality and gravity, making the force all zero.

Thanks for any help.

yuvaltassa, 2022-07-19:
The force sensor should work fine. If you add a static (doff-less) child body to the end of your arm, call this body "gripper", and then the constraint is between the gripper and the box, then you will see forces.

The force will certainly not equal gravity though. Maybe after you lift it and wait a long time for it to settle (probably add some fluid density so it stops swinging), then it'll equal gravity, but not before.

yinzikang, 2022-07-20:
Thanks for your reply!
I changed my XML file for a simple test according to it but still failed.
I think I should upload some figs and the code.
[image]

```
<worldbody>
        <light name="light" mode="targetbody" target="last link" directional="true" pos="1 1 3" />
        <geom name='floor' material="floor_mat" pos='0 0 0' size='2.5 2.5 0.1'  type='plane'/>
        <!--robot-->
        <body name="last link" pos="0 0 0.8">
            <inertial pos="0 0 0" mass="1" diaginertia="0.1 0.1 0.1" />
            <geom type="box" pos="0 0 0" size="0.1 0.1 0.1" rgba="1 0 0 1"/>
            <joint type="slide" axis="1 0 0"/>
            <body name="gripper" pos="0 0 -0.2">
                <inertial pos="0 0 0" mass="1" diaginertia="0 0 0" />
                <geom type="box" pos="0 0 0" size="0.1 0.1 0.1" rgba="0 1 0 1"/>
                <site name="sensor" pos="0 0 -0.1" rgba="1 0 0 0" size="0.01 0.01 0.01" type="box"/>
            </body>
        </body>
        <!--object-->
        <body name="box" pos="0 0 0.3">
            <inertial pos="0 0 0" mass="4" diaginertia="0.1 0.1 0.1" />
            <geom type="box" size="0.1 0.1 0.1"  rgba="0 0 1 1"/>
            <joint type="free"/>
        </body>
    </worldbody>
    <equality>
        <connect name="test" active="true" body1="gripper" body2="box" anchor="0 0 0"/>
    </equality>
    <sensor>
        <force name="force" site="sensor"/>
    </sensor>
```
The green `gripper` body is attached to the red `last link` body without joint, and the blue 'box' is constrained to the `gripper` with the `connect`.
The `sensor` site is defined below the `gripper` body which would calculate the force between the `last link` and the `gripper` according to the document.

When I start the simulation, sensor data equals `0 0 9.81` which is the gravity of the `gripper`. Shouldn't it be `0 0 5*9.81` for the gravity of the `gripper`(1 kg) and the `box`(4 kg)?

yuvaltassa, 2022-07-29:
Sorry, loading your own example model I'm not getting what you're reporting.

When I load the model below I see what you expect to see, the force reports the weight of all the bodies. This is your model but I had to fix it to make it loadable (add missing `<mujoco>`, delete undefined material), please only post loadable models. I also set gravity to -10, to make numbers round. Of course at the initial step the reported weight is not exactly 50, a bit of time needs to pass for the body to try to fall and for the equality to "catch" it and stabilise, but it is not far from 50 even at t=0.

<details>
  <summary>minimal XML</summary>
  
  ```xml
<mujoco>
    <option gravity="0 0 -10"/>

    <worldbody>
      <light name="light" mode="targetbody" target="last link" directional="true" pos="1 1 3" />
      <geom name='floor' pos='0 0 0' size='2.5 2.5 0.1'  type='plane'/>
      <!--robot-->
      <body name="last link" pos="0 0 0.8">
          <inertial pos="0 0 0" mass="1" diaginertia="0.1 0.1 0.1" />
          <geom type="box" pos="0 0 0" size="0.1 0.1 0.1" rgba="1 0 0 1"/>
          <joint type="slide" axis="1 0 0"/>
          <body name="gripper" pos="0 0 -0.2">
              <inertial pos="0 0 0" mass="1" diaginertia="0 0 0" />
              <geom type="box" pos="0 0 0" size="0.1 0.1 0.1" rgba="0 1 0 1"/>
              <site name="sensor" pos="0 0 -0.1" rgba="1 0 0 0" size="0.01 0.01 0.01" type="box"/>
          </body>
      </body>
      <!--object-->
      <body name="box" pos="0 0 0.3">
          <inertial pos="0 0 0" mass="4" diaginertia="0.1 0.1 0.1" />
          <geom type="box" size="0.1 0.1 0.1"  rgba="0 0 1 1"/>
          <joint type="free"/>
      </body>
    </worldbody>

    <equality>
        <connect name="test" active="true" body1="gripper" body2="box" anchor="0 0 0"/>
    </equality>

    <sensor>
        <force name="force" site="sensor"/>
    </sensor>
</mujoco>
```

</details>

closed 2022-07-31 by copybara-service[bot]

yuvaltassa, 2022-07-31:
We've now added this feature, see [documentation](https://mujoco.readthedocs.io/en/latest/XMLreference.html#adhesion) and [video](https://youtu.be/HdBue4MUZys).

Currently only active contacts can generate an adhesive force. It is possible to generalise this to non active ("excluded") contacts, which would give the action-at-a-small-distance effect of vacuum grippers. Please write here if you think you need this.

Cheers

Lucky-hqj, 2022-08-01:
This looks very cool ! ! !   Will there be a release for this feature?

yuvaltassa, 2022-08-01:
The next release will include the feature (some time in the next few weeks). In the meantime feel free to build MuJoCo yourself to try it out.

yinzikang, 2022-08-06:
> Sorry, loading your own example model I'm not getting what you're reporting.
> 
> When I load the model below I see what you expect to see, the force reports the weight of all the bodies. This is your model but I had to fix it to make it loadable (add missing `<mujoco>`, delete undefined material), please only post loadable models. I also set gravity to -10, to make numbers round. Of course at the initial step the reported weight is not exactly 50, a bit of time needs to pass for the body to try to fall and for the equality to "catch" it and stabilise, but it is not far from 50 even at t=0.
> 
> minimal XML

Sorry for my late response and unloadable XML.
I tried again with the latest mujoco and the problem got solved somehow. Maybe it is a version-related problem or something.
At last, thank you so much for your test and your new features.

Best wishes.

yuvaltassa, 2022-08-19:
Quick update, the adhesion actuators now support _excluded contacts_ which effectively let you have a layer of action-at-a-distance, like a vacuum gripper. See the updated video, documentation and example model [here](https://mujoco.readthedocs.io/en/latest/XMLreference.html#adhesion)

commit 3d77eb1e (2022-07-31): Add adhesion actuators.

- Adhesion actuators using contact normals as force transmission mechanism.
- Related video: https://youtu.be/HdBue4MUZys

Closes #229

PiperOrigin-RevId: 464389367
Change-Id: I9f69b3cd152d957e8f65870d208788463c036a6d