google-deepmind/mujoco #231: RangeUpper issue with BallJoint in Unity Plugin

Balint-H, 2022-04-08:
When importing an XML with a ball joint with limited range defined in degrees, the range value is converted to radians in the Unity component. The property hint itself is defined like that as well. However, when running the scene the actual limits are the degree "equivalent" of the radian range of motion (i.e., to get 90 degree range of motion, RangeUpper needs to be set to 90 "radians").

This is on Windows 10, Unity 2021.2.5, most recent version of the plugin.

Here's the XML I noticed this in:

```
<mujoco model="Ball">   
    <compiler angle="degree" meshdir="mesh/" texturedir="texture/"/>
    <option timestep="0.002" iterations="50" apirate="50" solver="Newton" tolerance="1e-10" cone="elliptic"/>
    <size njmax="600" nconmax="150" nstack="300000"/>
	
	<default>
	    <default class="free">
	        <joint type="free" damping="0" armature="0" limited="false"/>
	    </default> 
		<joint type="hinge" damping=".2" stiffness="1" armature=".01" limited="true" solimplimit="0 .99 .01"/>
		<geom type="capsule" condim="1" friction=".7" solimp=".9 .99 .003" solref=".015 1"/>
	</default> 
	
    <worldbody>
	<geom type="plane" pos="0 0 0" size="1.5 1.5 .025"/>
		
        <body name="mocap1" mocap="true" pos="-1 0 1" axisangle="0 0 1 3.141592">
            <geom type="box" group="1" size="0.02 0.02 0.02" contype="0" conaffinity="0" rgba=".9 .5 .5 1"/>
        </body>
		

		<body name="RightShoulder" pos="-1 0 1">
			<geom name="RightShoulderGeom" type="sphere" size="0.04"/>
			<joint class="free"/>
			<body name="RightArm" pos="0 0 0">
				<joint type="ball" name="right_shoulder" range="0 70" damping=".2" stiffness="1" armature=".01" limited="true" solimplimit="0 .99 .01"/>
				<geom name="RightArmGeom" fromto="0.04 0 0 .238 0 0" size="0.0398"/>
				<body name="RightForeArm" pos=".278 0 0">
					<joint name="right_elbow" axis="0 0 -1" range="-135 5" stiffness="0"/>
					<geom name="RightForeArmGeom" fromto=".04 0 0 .243 0 0" size=".04"/>
					<body name="RightHand" pos=".283 0 0">
						<geom name="RightHandGeom" type="box" pos="0.08 0 0" size=".08 .05 .0125"/>
					</body>
				</body>
			</body>
		</body>
    </worldbody>

	<actuator>
		<position name="right_shoulder1" group="1" gear="5 0 0"  joint="right_shoulder"/>
		<position name="right_shoulder2" group="1" gear="0 5 0"  joint="right_shoulder"/>
		<position name="right_shoulder3" group="1" gear="0 0 5"  joint="right_shoulder"/>
	</actuator>
	
	<actuator>
		<position name="right_elbow"     gear="2"  joint="right_elbow"/>
	</actuator>
	
	<equality>
        <weld body1="mocap1" body2="RightShoulder" solref="0.01 1" solimp=".9 .9 0.01"/>
	</equality>
</mujoco>
```

erez-tom, 2022-04-12:
You're right.  In [MjHingeJoint.cs](https://github.com/deepmind/mujoco/blob/main/unity/Runtime/Components/Joints/MjHingeJoint.cs) you can see the various accommodations we go through to ensure radian/angle interplay (e.g., `MjSceneImportSettings.AnglesInDegrees`), which are currently not used for the ball joint class.  We will amend this in the next version, but if this is blocking you feel free to send a pull request.

closed 2022-09-29 by Balint-H

joanllobera, 2025-02-18:
There seems to be a similar problem, in that when I import a ball joint where the upper limits are converted in radians, despite they are processed as degrees. For example, if the .xml file describes a ball joint like:
`          <joint name="lfemurr" type="ball" class="stiff_medium" solimplimit="0.9 0.95 0.001 0.5 2" range="0 137.51" />`

In the Unity inspector the field *Range Upper* appears like  2.4, despite the value is expected to be in degrees. 
The Unity exporter, though, seems to work well, the value in the inspector is the same value found in the .xml that is generated. 

Maybe this issue should be reopened?