google-deepmind/mujoco #232: what's the different of two mujoco and mujoco-py both from pip?

jinfagang, 2022-04-09:
It makes users very confused. Any explanation?

rohanpsingh, 2022-04-09:
mujoco-py were the python bindings of mujoco developed by openai (prior to acquisition by deepmind). I’m not sure if it is still actively maintained. Then, last month, developers at deepmind released the python bindings for mujoco with version 2.1.2.

The purpose of both is the same. But `pip install mujoco` is officially supported. So my opinion is that for most users this would be the recommended choice.

jinfagang, 2022-04-09:
@rohanpsingh Previous mujoco-py provides very impressive and rich examples, I can not see any from mujoco official python bindings, will these part can merge gap between them? I can not found any tutorials or simple examples using mujoco official python bindings, it was just a package, and all with python bindings even hard to found pure python code to take a deep inside look into implementation.

saran-t, 2022-04-09:
The `mujoco` package aims to provide direct, low level bindings for MuJoCo and tries to adhere to its C API as closely as possible. The functionality provided in [the `python` directory of our repo](https://github.com/deepmind/mujoco/tree/main/python) is roughly equivalent to what's implemented in the [`pxd` directory for `mujoco-py`](https://github.com/openai/mujoco-py/tree/master/mujoco_py/pxd). These bindings are implemented in C++ through pybind11.

Basic example usage can be found in [bindings_test.py](https://github.com/deepmind/mujoco/blob/main/python/mujoco/bindings_test.py), although admittedly that's not exactly a user-friendly example. We have also [retrofitted our `dm_control` RL library](https://github.com/deepmind/dm_control/commit/0425d679046baab5772dffd86a5af1cff49f8502) to use the new bindings, and for higher-level use cases we would suggest that you look there too. Separately, there is an ongoing effort to [migrate `gym` to use our new `mujoco` bindings](https://github.com/openai/gym/pull/2595).

The new bindings is slightly less than one month old at this point and it'll take a bit of time for us to build a library of documentation and examples. Having said that, our decision to make the `mujoco` Python package correlate closely to MuJoCo's C API means that people should be able to reference [MuJoCo's existing documentation](https://mujoco.readthedocs.io/en/latest/) in most cases. The [Python chapter](https://mujoco.readthedocs.io/en/latest/python.html) there also provides more detail into the bindings and some rough guide for how to migrate from `mujoco-py`. (Again, we realise that our Python documentation needs to be improved -- that's work in progress.)

In the meantime, if you run into any issue that's not covered by our docs, please feel free to reach out to us via GitHub issues and we'll try to help you out.

jinfagang, 2022-04-10:
@saran-t thanks, I am very glad to see there is a official new mujoco python package. Old one is out-dated with newest mujoco installation software, this way looks good, so that we can using python bindings to more focus on control and algorithm design.

I have an old project which takes a `xml` model as target object, I want make it move, where should I started with the new package?

[image]

(am pretty new in mujoco)

saran-t, 2022-04-12:
When you say you want to make it move, do you mean with random control signals that just causes the humanoid to wriggle around on the ground (e.g. to sanity check your model), or to get the humanoid to do something useful like walking?

If it's just random movement that you're after, you need something that generates a random walk sequence and sets it to the `ctrl` field in `mjData`. See https://github.com/deepmind/mujoco/blob/main/sample/testspeed.cc for an example (that's in C++ but you should be able to translate the core algorithm into Python fairly straightforwardly).

If you want meaningful locomotion, you'll need to make your own controller (e.g. by training one using reinforcement learning, or model based optimisation).

AmetistDrake, 2022-04-27:
@jinfagang I would recommend you to use the `mjcb_control` callback function, and inside the callback set the mjData.ctrl field (unlike in the `testspeed.cc` example, where it is directly accessed without the callback). This way your control data will be applied at the right time during the simulation pipeline. For details why this is the recommended way see the `Simulation loop` part of the docs.  [https://mujoco.readthedocs.io/en/latest/programming.html#simulation-loop](url)