google-deepmind/mujoco #234: How to constrain one heirachy of bodies to a seperate body?

livinamuk, 2022-04-10:
Is it possible in MuJoCo to constrain one heirachy of bodies to a seperate body?

For example, a humaoid ragdoll attached to a separate static body?

[image]

yuvaltassa, 2022-04-10:
Not clear what you are asking. Constrain how?

Are you just asking if MuJoCo has [tendons](https://mujoco.readthedocs.io/en/latest/XMLreference.html#tendon)? The answer to question is yes.

livinamuk, 2022-04-10:
My understanding of MuJoco is that joints are created between two bodies, a parent and its child. Is it possible to create a joint between two bodies which do not share a parent/child relationship?

In the screenshot example this would be between the hand of the ragdoll and a separate body, one not part of the ragdoll heirachy.

yuvaltassa, 2022-04-10:
MuJoCo has both spatial tendons and equality constraints. One or both of those might be what you are looking for. Please read the related documentation.

closed 2022-04-10 by yuvaltassa