google-deepmind/mujoco #239: Object slips from end-effector

chisarie, 2022-04-13:
Hi, 

I have a franka emika panda model and I wrote a short script to make it pick up a small cube. The problem is that this cube slips away from the end-effector, see the attached video. I tried playing around with some friction parameters, but I did not manage to get it work. Do you have any suggestions?

The cube is modeled as 
```xml
<body name="cube" pos="1 1 1">
    <joint name="cube_joint" type="free" damping="1"></joint>
    <geom type="box" size=".02 .02 .02" mass="0.1" rgba="0 0.7 0 1.0" />
</body>
```

Thank you very much in advance!

https://user-images.githubusercontent.com/43339851/163166523-a234ba28-67c5-4ccb-bab1-ff1f6df9bdaa.mp4

yuvaltassa, 2022-04-13:
What is the colliding geom on the tip of your end-effector?

If it is the actual mesh I would strongly recommend to disable the collisions for the mesh and add a layer of (possibly invisible) box primitives that overlap with the end effector.

chisarie, 2022-04-14:
Hi Yuval,

thank you very much for the tip. Yes, the colliding geom was the actual mesh. I now changed it to box primitives as suggested and the result is already much better. 
Nevertheless, there is still a slow slipping happening, which means that the block is still falling from the gripper, just more slowly.
I tried playing with friction on both the object and the primitive gripper boxes by setting for example `friction="2 0.005 0.0001"`, but this resulted in very bad penetration, i.e. the gripper just went through the object.

Is there any other way I can make the object or the gripper box primitive "stickier"?

Thank you again!

yuvaltassa, 2022-04-14:
Happy to help! Read [here](https://mujoco.readthedocs.io/en/latest/overview.html#softness-and-slip) for a discussion of this subject.

closed 2022-04-14 by yuvaltassa

QiyangYan, 2023-09-07:
Hi @yuvaltassa, thanks for your tips but I have a follow-up question regards. My mesh file is a hexagonal prism and instead of having multiple boxes inside to for every surface of the object, is there an easier way?