google-deepmind/mujoco #243: Reference frame of the angle speed of the floating object

yinzikang, 2022-04-14:
I am trying to simulate the movement of a floating box linked to the "worldbody" with a free joint, and I checked the document which told me that as for the floating object, the "qvel" controls the speed of the box with the world as the reference system
The linear velocity works well even when the box is rotated, and it moves along the world coordinate system
However, when I add some angular velocity to the box, the box rotates with its own coordinate system, which is not expected.
So I wonder is there something wrong with my understanding or else.
Here is my code, thanks for any help
[image]
[image]

yuvaltassa, 2022-04-14:
Your understanding is correct, but I can elaborate:
- The _position_ of a free joint is in the global frame and so are linear velocities due to the trivial differentiation of linear quantities.
- The _orientation_ of a free joint is also with respect to the global frame, given by a unit quaternion. However the tangent-space of quaternions is not trivial and indeed rotates with the quaternion. The upshot is that rotational _velocities_, which live in the tangent space, are effectively in the _local frame_, even though the orientation itself is in the global frame.

You can read more about unit quaternions [here](https://en.wikipedia.org/wiki/Quaternions_and_spatial_rotation) and elsewhere.

Your post leads me to think that perhaps there is a bit of confusing documentation somewhere? If yes please let us know where it is so we can clarify.

closed 2022-04-14 by yuvaltassa

yinzikang, 2022-04-14:
Oh, I see, thanks so much. 
The "Floating objects" part in https://mujoco.readthedocs.io/en/latest/overview.html#floating-objects always mentions the two speeds together without information about the difference in their frame, which makes me think they relate to the same one. Or maybe I am the only one who gets confused, lol.

yuvaltassa, 2022-04-14:
You are definitely not the only one! 
[SO3](https://en.wikipedia.org/wiki/3D_rotation_group) is a confusing thing.
Thank you for the pointer, we will improve this documentation section to better address this common question.