google-deepmind/mujoco #244: Python bindings: wrong signature in demo example

ikamensh, 2022-04-15:
Hello,

### Problem
in the [intro page of python bindings](https://mujoco.readthedocs.io/en/latest/python.html#minimal-example), example has following code:
`model = mujoco.MjModel.from_xml_string(XML, ASSETS)`

It triggers a warning in pycharm type checker: "Parameter 'assets' unfilled", because as of mujoco 2.1.5 the type hint for `from_xml_string` thinks it's not a static method:
`def from_xml_string(self, xml, assets, Dict=None, p_str=None, bytes=None, *args, **kwargs): # real signature unknown; NOTE: unreliably restored from __doc__ `
 
### Motivation for fixing this
To avoid bugs in the code, I want to use type checkers with mujoco bindings. 
----
Is there a way to ship correct signatures?

saran-t, 2022-04-21:
We've had mixed success with auto-generated type stubs for internal usage. The `.pyi` stubs were generated using [mypy](https://github.com/python/mypy), but the tool gets confused in a few places and we have a separate script that performs specific textual replacements on the output.

Perhaps try mypy and let us know how well that works for you?

ikamensh, 2022-04-22:
Not sure I've installed it right, but it fails to find stubs for mujoco:

```
(venv) ~/P/Mujee $ mypy -c "import mujoco"                                                                                                         
<string>:1: error: Cannot find implementation or library stub for module named "mujoco"
<string>:1: note: See https://mypy.readthedocs.io/en/stable/running_mypy.html#missing-imports
Found 1 error in 1 file (checked 1 source file)
```

I also don't see how mypy can accept signature containing self (non-static method) being used as a static method.

ikamensh, 2022-04-22:
if you intend for type checkers to use sources for types, you should include `py.typed` file in the source tree. https://blog.whtsky.me/tech/2021/dont-forget-py.typed-for-your-typed-python-package/

horizon-blue, 2026-10-16:
🤔 I wonder if there's any plan for MuJoCo to officially ship the type stubs along with the package? That way, people don't have to generate & maintain their own copies and keep them up-to-date as they update MuJoCo. Alternatively, there's also a route similar to [scipy-stubs](https://github.com/scipy/scipy-stubs) which ships the type stubs as a sibling package following [PEP561](https://peps.python.org/pep-0561/#stub-only-packages), so people can help contribute and fix it without touching the main MuJoCo codebase.

In the mean time, if people are looking for solutions, I had pushed a set of unofficial MuJoCo type stubs to [mujoco-stubs package on PyPI](https://pypi.org/project/mujoco-stubs/), which should cover MuJoCo 3.4.0-3.16.0:

```bash
pip install "mujoco-stubs==3.16.0.*" # or replace with your matching MuJoCo version
```

The stubs were generated using something similar to https://github.com/google-deepmind/mujoco/issues/1292#issuecomment-3208219200 plus manual fixes. I took the `mujoco-stubs` package name to adhere to [PEP561](https://peps.python.org/pep-0561/#stub-only-packages). If the MuJoCo team want to take over this package name and start maintaining the type stubs officially, lmk! I will we be more than happy if there's an official route to the type checking experience :).