google-deepmind/mujoco #246: Semi-Transparency Bug in Visualizer

kevinzakka, 2022-04-17:
A bug for visualizing semi-transparent geoms seems to have been introduced in `mujoco` versions >= 2.1.3.

|   2.1.2  |  2.1.3/4/5   |
|---|---|
|  [image]   | [image]|

Note I have not selected any geom in the viewer. The forearm stays opaque in newer versions.

yuvaltassa, 2022-04-19:
Hi Kevin!

This is intentional. As you can see in item 2 of the [2.1.3 changelog](https://mujoco.readthedocs.io/en/latest/changelog.html#version-2-1-3-mar-23-2022) (though this can probably be explained better), the [mjVIS_STATIC](https://mujoco.readthedocs.io/en/latest/APIreference.html#mjtvisflag) category used to include only direct children of the world, and now includes any dof-less descendants. Because your forearm is a static, it now stays opaque, just like the floor stays opaque. Try 'D' to toggle all static bodies.

That said, I think we'll make an exception for mocap bodies, which are technically dof-less descendants, but can still be moved. Your forearm is not a mocap body, but it probably should be. It is currently a child of an otherwise empty body called "shadow_hand_e/" which AFAICT is only there to rotate it? I would recommend removing that body, changing the frame of the forearm directly, and making it a mocap body. In the next release mocap bodies transparency will be toggleable.

closed 2022-04-19 by yuvaltassa

kevinzakka, 2022-04-19:
Hi @yuvaltassa, appreciate your response as always!

Ended up going the `mocap` body + weld constraint approach, as documented [here](https://mujoco.readthedocs.io/en/latest/modeling.html#motion-capture). The steps I took were to:

1. Add a `geom`-less mocap body to the arena worldbody.
2. Make the forearm body position and quaternion equal to that of the mocap body.
3. Add the hand to the arena as a free-floating body (i.e., by adding a free joint).
4. Add a weld equality constraint between the mocap body and the forearm.

My hand now floats in space and becomes transparent when I press `T`. Additionally, because the mocap body does not have a geom, I don't have to worry about it remaining visible after `T` is pressed.

Cheers