google-deepmind/mujoco #248: Error occured while trying connect rigid body and rope

woshizh951, 2022-04-21:
Hi 

I want to bulid a simulation system that one end of cable connceted with robot end-effector and another end connceted with a fixed box.
I have generate cable and box in mujoco, and succeed in connect cable to the effector.
However, when i'm trying to connect another end of cable to the box, the connection was builld successful but the end connected with a point with nothing there and away from box.   
[image]

you can find the main script of the env. I connect object by equality-connect function, current order of body 1 and body2 is the only avaiable order i tested, the cable would be crazy as long as i change the connect order.
 I guess the reason is wrong anchor value? but when i change the anchor pos, system become unstable.

[123.txt](https://github.com/deepmind/mujoco/files/8526845/123.txt)

Could you please give any suggestions?

Best Regards.

yuvaltassa, 2022-04-21:
Hi,

Your model does not load. Please attach a loadable model (as a zip file, including all assets) if the suggestion below does not solve you problem.

From inspection of your image, it seems like possibly your constrained points do not overlap at `qpos0`? As described [here](https://mujoco.readthedocs.io/en/latest/computation.html#coequality), connect constraints are by default defined with respect to the null configuration of your model (`model->qpos0`). If this is the case, you would need to rotate your rope so that the two constrained points overlap at the null pose. This is of course just a guess since I cannot load your model.

woshizh951, 2022-04-21:
Hi , 

I saw you said the qpos0, but i don't know how to set it in xml file. Should I change it by mujoco-py?

I have been submit the loadable model. you can find it in zip file and run the connect_by_rope.xml to recurring the error.
could you please help me to see any problems cause the unwilling result?
[for update.zip](https://github.com/deepmind/mujoco/files/8529654/for.update.zip)

Also, I'm trying to connect the box and robot ee by tendon, It succeed, the tendon is attached like what i want. Does the cable can be like this?  you can also see the tendon-connected case by running conect_by_tendon.xml.

[image]

Best Regards.

yuvaltassa, 2022-04-21:
A tendon is a good solution if you are not interested in modeling the dynamics of the cable.

Regarding `qpos0` this is simply the configuration that you see when you first load the model. Right now your rope is pointed sideways. I should point down, so that the tip touches the point where you want to connect it.

Also, I noticed you gave your rope a free joint. It will probably be more stable if you made it a child of the robot or of the object.

closed 2022-04-21 by yuvaltassa