google-deepmind/mujoco #254: XML File Extension Letter Case Behavior Different macOS vs. Linux

kevinzakka, 2022-04-25:
MuJoCo will fail to load an XML file on Linux if the file extensions defined in the XML file do not match the letter casing of the actual file on disk. This makes sense given Linux is case-sensitive but now we get diverging functionality on Linux and macOS. Ideally, MuJoCo should check for both lower and upper case extensions when compiling the XML file.

Here's a minimal ZIP (XML + asset) to reproduce this issue: [test_mjcf.zip](https://github.com/deepmind/mujoco/files/8558232/test_mjcf.zip)

saran-t, 2022-04-25:
I'm not convinced that's a good idea actually. It can be surprising for people who are using case-sensitive filesystems who would rightly expect the parser to respect that.

I think it's OK for us to require that XMLs refer to filenames that exist, according to whatever "exists" means on the filesystem in question. Unless you're using something like FAT32, filesystems that aren't sensitive to cases nevertheless _preserve_ cases.

kevinzakka, 2022-04-25:
That makes sense, I guess it just causes friction for people sharing assets and models. FYI, this happened when I downloaded someone's model from the old mujoco forum -- thinks worked great on my mac but CI tests kept failing on Github and it took me a while to figure out what the issue was.

kevinzakka, 2022-04-25:
Thinking about it some more, I don't think a typical MuJoCo user has both a `foo.STL` and `foo.stl` and interchanges between them in different parts of a model.

kevinzakka, 2022-04-26:
@saran-t Maybe we can modify dm_control's MJCF parser and make it output a warning if it sees this discrepancy?

kevinzakka, 2022-04-26:
@saran-t If the preference is not to make assumptions at the MuJoCo level, then can we try to be more helpful at the PyMJCF level? I've made an attempt over at `dm_control`, would be great to get your thoughts on that!

saran-t, 2022-04-28:
We've discussed this within the team and think that the best way to issue a warning would be to list all files under the directory in question, go through each one and check for string equality with both operand fully lowercased.

kevinzakka, 2022-04-28:
Sounds great, thanks @saran-t !!

closed 2023-10-16 by kevinzakka