google-deepmind/mujoco #256: the unlocked activation key returns "ERROR: Expired activation key"

zhihaocheng, 2022-04-27:
I employ the unlocked activation key to install mujoco200 on my school's HPC(High Performance Computing). However, it returns "ERROR: Expired activation key". This problem is quite strange because I am sure the right activation key is used. The same problem for installing mjpro150. Could you please help me?

saran-t, 2022-04-27:
Are you using the activation file from https://roboti.us/license.html ?

Separately, do you have a specific reason for using these legacy versions of MuJoCo?

zhihaocheng, 2022-04-27:
> Are you using the activation file from https://roboti.us/license.html ?
> 
> Separately, do you have a specific reason for using these legacy versions of MuJoCo?

Thanks. Yes, of course. I downloaded the activation file from the url as you provide. This is quite a strange problem because this activation works on my own computer but not the HPC of school. 

The reason for using the legacy version is that my previous experiments were run with the old version. I will test whether the new Mujoco work for me.

saran-t, 2022-04-27:
Are you able to double check if the clock on the cluster is set correctly?

zhihaocheng, 2022-04-28:
@saran-t, Thanks Saran. I have tested the new version, and it works. For the system clock, I run command "date" and receive correct time. For the hardware clock, I have written to the cluster administrators to check.

saran-t, 2022-04-28:
Thanks for the info. We don't have visibility on how exactly the clock is being used in `mj_activate` as all license management code has been removed from the MuJoCo codebase by Roboti before it has been transferred to us. We will follow up with Roboti on this.

nimrod-gileadi, 2022-05-19:
The summary of our discussion with Roboti is that MuJoCo prior to 2.1.0 had some logic to detect tampering with licenses. This included various check to detect the system clock being set to an old date, including detecting files with modification timestamps in the future.

This is no longer an issue with new versions of MuJoCo, of course.

closed 2022-05-19 by nimrod-gileadi