google-deepmind/mujoco #257: unable to load a simple model file

jinfagang, 2022-04-27:
```
    model = mujoco.MjModel.from_xml_string('/home/mujoco/model/humanoid/humanoid.xml', ASSETS)
ValueError: XML parse error 8:
Error=XML_ERROR_PARSING_TEXT ErrorID=8 (0x8) Line number=1

```

the model file are come from here:

[image]

my code:

```

ASSETS=dict()

model = mujoco.MjModel.from_xml_string('/home/mujoco/model/humanoid/humanoid.xml', ASSETS)
data = mujoco.MjData(model)

while data.time < 1:
    mujoco.mj_step(model, data)
    print(data.geom_xpos)
```

saran-t, 2022-04-27:
The error says that there's something wrong with the XML file. You need to open up `'/home/mujoco/model/humanoid/humanoid.xml'` and see what's in it.

jinfagang, 2022-04-27:
@saran-t Hi, it was simple this file: https://github.com/deepmind/mujoco/blob/main/model/humanoid/humanoid.xml

anything wrong with the file you provided?

yuvaltassa, 2022-04-27:
`from_xml_string` [takes the actual XML string](https://github.com/deepmind/mujoco/blob/main/python/mujoco/bindings_test.py#L84), not the filename. Try loading the string yourself or using the `from_xml_path` constructor.

jinfagang, 2022-04-27:
@yuvaltassa OH......!

jinfagang, 2022-04-27:
@yuvaltassa BTW, how to make Mujoco able to promot on vscode? I can not get any code jump with mujoco...(python)

yuvaltassa, 2022-04-27:
Sorry, "promot"?

jinfagang, 2022-04-28:
@yuvaltassa yes. code suggestions in IDE.