google-deepmind/mujoco #258: Extrinsic camera matrix from mujoco.MjvCamera()

Barbany, 2022-04-28:
I want to compute de extrinsic camera parameters (rotation and translation) from the attributes of an instance of the class `mujoco.MjvCamera()` so that I can project the points of the manipulated object from world to camera coordinates:
`point_cam_coords = R @ point_world_coords + t`

The `camera` location and rotations are described by its attributes `azimuth`, `lookat`, `distance`, and `elevation`.

The rotation matrix `R` can be computed using `camera.azimuth` and `camera.elevation` but it is not clear to me how to do so (essentially in which order the rotations are applied and if previous transformations were applied, e.g., I assume the z-axis is rotated by 180 degrees at the end to compensate the inverted figure).

Then the translation vector can be easily computing by enforcing that `camera.lookat` point (expressed in world coordinates) is translated to the vector `[0, 0, camera.distance]` in camera coordinates (z-axis pointing towards it). This can be done by letting
`t = np.array([0, 0, camera.distance]) - R @ camera.lookat`.

Can anyone help with this?

Barbany, 2022-04-28:
I think I got it! First apply azimuth rotation along the z-axis, then apply a rotation along the x-axis composed of the elevation and 90 degrees shift (we want z-axis to be pointing to the object). Finally apply 180 degrees rotation along the z-axis (image is upside down).
[image]

```
R = Rz(np.pi) @ Rx(np.radians(camera.elevation) + np.pi / 2) @ Rz(np.radians(camera.azimuth))
t = (np.array([0, 0, camera.distance]) - R @ camera.lookat).reshape(-1, 1)
```

I'm closing the issue and leaving this in case someone encounters the same problem.

yuvaltassa, 2022-04-28:
[This is how we do it](https://github.com/deepmind/dm_control/blob/main/dm_control/mujoco/engine.py#L700), but our code uses the Cartesian scene camera directly, after the lookat-distance-azimuth-elevation transform has already been applied. You can also check out the cell titled "Projecting from world to camera coordinates" in our [tutorial colab](https://github.com/deepmind/dm_control/blob/main/dm_control/mujoco/tutorial.ipynb).

yuvaltassa, 2022-04-28:
The [associated test](https://github.com/deepmind/dm_control/blob/main/dm_control/mujoco/engine_test.py#L191) might also be helpful.

closed 2022-04-28 by yuvaltassa

Barbany, 2022-04-28:
My problem was actually with `mujoco_py`, where the scene (and hence the x matrix and pose) is not updated after the camera attributes are changed. That's why I asked to only use the camera attributes and not the scene as you do in the code you pointed me to.

Using your Python bindings, the method to update the scene has to be explicitly called so in this case, I could follow your tutorial.