google-deepmind/mujoco #259: illegal hardware instruction on ubuntu 18.04

Trinkle23897, 2022-04-29:
```bash
$ pip3 install mujoco==2.1.5
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: mujoco==2.1.5 in /home/jiayi/.local/lib/python3.8/site-packages (2.1.5)
Requirement already satisfied: absl-py in /home/jiayi/.local/lib/python3.8/site-packages (from mujoco==2.1.5) (1.0.0)
Requirement already satisfied: glfw in /home/jiayi/.local/lib/python3.8/site-packages (from mujoco==2.1.5) (2.5.3)
Requirement already satisfied: numpy in /home/jiayi/.local/lib/python3.8/site-packages (from mujoco==2.1.5) (1.22.3)
Requirement already satisfied: pyopengl in /usr/lib/python3/dist-packages (from mujoco==2.1.5) (3.1.0)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from absl-py->mujoco==2.1.5) (1.11.0)

$ python3
Python 3.8.0 (default, Dec  9 2021, 17:53:27)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco
[1]    16930 illegal hardware instruction (core dumped)  python3

$ cat /proc/cpuinfo | grep name | uniq -c
     24 model name      : Intel(R) Xeon(R) CPU           E5645  @ 2.40GHz

$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.5 LTS"
```

I can provide more information if needed. BTW, is it possible to provide `.a` file instead of `.so`?

saran-t, 2022-04-29:
This is because your CPU [doesn't support AVX](https://ark.intel.com/content/www/us/en/ark/products/48768/intel-xeon-processor-e5645-12m-cache-2-40-ghz-5-86-gts-intel-qpi.html). We're close enough to open sourcing now that I don't think we'll do anything about this now, but once you have source access you should be able to build a non-AVX (and indeed a static library, if needed) yourself.

Could you tell us why you need a static rather than dynamic library? From our point of view it's much easier to ship things to people as a DSO since we can bundle known-good versions of MuJoCo's dependencies and make sure that we correctly hide the symbols from our deps so that they can coexist with other versions of the same libraries if the user ends up pulling them in (e.g. through scipy).

yuvaltassa, 2022-04-29:
Re timeline, see [here](https://github.com/deepmind/mujoco/discussions/126#discussioncomment-2600655).

closed 2022-04-29 by yuvaltassa