google-deepmind/mujoco #262: Mujoco Fortran Interface

general-rishkin, 2022-05-06:
I have written a Fortran interface to Mujoco using the C API but it is not working properly. I cannot see the problem so I hope I can get some help to review and diagnose the issue. The Fortran binding is [here](https://github.com/general-rishkin/mujoco-fort). It is an interface I want to maintain using Modern Fortran.

The problem is that on running a simple test, I get a blank screen and then a ‘not responding’ pop up a few seconds later, like so:
[image]

What I should get is:
[image]

To reproduce the problem :
- The src folder consists of the Fortran interfaces to the C api (the C header files are in src/c_api). So, for example, the file src/mod_mjmodel corresponds to the file src/c_api/mjmodel.h.
- The relevant Mujoco libraries are placed in the folder mujoco_libs. So, all you need to do is ensure they can be found when building and of cousre, you will also require libGL and libglfw.
- In line 52 in app/main_mujoco_test.f90, you need to replace the:

>  “CHANGE_THIS_TO_PATH_TO_BALL.XML_IN_SRC/ASSETS_FOLDER/ball.xml”

with the path to ball.xml, which is in the folder src/assets/ball.xml .

- Then just run with: fpm run test_mujoco.   
fpm is the [Fortran Package Manager](https://github.com/fortran-lang/fpm)

I was not certain if this should go here or in 'Discussions', so my apologies if I have posted this in the wrong section. 
Thanks in advance.

general-rishkin, 2022-05-06:
Thanks, this is great advice. I will try to write a Fortran equivalent of `testspeed.cc`.

saran-t, 2022-07-15:
I'm going to close this issue since it's been dormant for a while. Feel free to reopen if you have any update on the topic though.

closed 2022-07-15 by saran-t