google-deepmind/mujoco #263: How to wind the rope around the winch?

Zcyyy, 2022-05-06:
Hi:
 I want to know how to configure the XML file to wrap the rope around a post or winch, something like this.
[image]
A rope wrap around a post.

nimrod-gileadi, 2022-05-08:
If you want to simulate the physics of the rope, you can attach one end of the rope to the winch cylinder with a ball joint or an equality constraint.
To simulate the winch you would use a hinge joint, to which you can attach an actuator.
e.g, based on the existing [rope example model](https://github.com/deepmind/mujoco/blob/main/model/composite/rope.xml):

```
<mujoco model="winch">
  <option timestep="0.0004" jacobian="dense"/>
  <size nconmax="500" njmax="1500" nstack="1000000"/>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512"/>
    <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2"
      width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>

    <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
  </asset>

  <worldbody>
    <geom name="ground" type="plane" size="0 0 1" material="matplane"/>
    <body name="winch" pos="0 0 0.13">
      <joint name="winch"/>
      <geom type="cylinder" size="0.1 0.1"/>
      <geom type="cylinder" size="0.2 0.01" pos="0 0 -0.11"/>
      <body name="B0" pos="0.11 0 0">
        <joint type="ball"/>
        <composite type="rope" count="51 1 1" spacing="0.06" offset="0 0 2">
          <joint kind="main" damping="0.005"/>
          <geom type="capsule" size=".01 .025" rgba=".8 .2 .1 1"/>
        </composite>
      </body>
    </body>
  </worldbody>
  <actuator>
    <velocity name="winch" joint="winch" kv="200" ctrllimited="true" ctrlrange="-10 10"/>
  </actuator>
</mujoco>
```

If you want a more efficient simulation of a taught rope as part of a mechanism, you could use a tendon.

closed 2022-05-08 by nimrod-gileadi

Zcyyy, 2022-05-09:
Yes, I have implemented what you described. I think I didn't describe my needs clearly. I want one section of the rope to be attached to the winch, the rope to be wound onto the winch at least once, and the other end of the extended rope to be attached to another rigid body. To implement this function, we need to solve one of the following two problems: 1.Can the rope be wound directly onto the winch by editing the XML file? Or 2. Whether it can make a weld or a connection between the end of the rope and a rigid body in the program. (Realize the connection of objects in space. If WELD or CONNECT is written to the XML file, the rope will always make a virtual connection with the rigid body in its initial position).

Zcyyy, 2022-05-09:
Yes, I have implemented what you described. I think I didn't describe my needs clearly. I want one section of the rope to be attached to the winch, the rope to be wound onto the winch at least once, and the other end of the extended rope to be attached to another rigid body. To implement this function, we need to solve one of the following two problems: 1.Can the rope be wound directly onto the winch by editing the XML file? Or 2. Whether it can make a weld or a connection between the end of the rope and a rigid body in the program. (Realize the connection of objects in space. If WELD or CONNECT is written to the XML file, the rope will always make a virtual connection with the rigid body in its initial position). @nimrod-gileadi

Zcyyy, 2022-05-09:
@nimrod-gileadi  Thanks a lot.
Yes, I have implemented what you described. I think I didn't describe my needs clearly. I want one section of the rope to be attached to the winch, the rope to be wound onto the winch at least once, and the other end of the extended rope to be attached to another rigid body. To implement this function, we need to solve one of the following two problems: 1.Can the rope be wound directly onto the winch by editing the XML file? Or 2. Whether it can make a weld or a connection between the end of the rope and a rigid body in the program. (Realize the connection of objects in space. If WELD or CONNECT is written to the XML file, the rope will always make a virtual connection with the rigid body in its initial position).

yuvaltassa, 2022-05-09:
1. It sounds like you want the rope to be wound onto the winch at the starting position, is that correct? If yes, why? Are you familiar with the ability to save keyframes into the XML? That would be the easiest thing to do. Winding the rope in the initial pose is not easy.
2. Yes, making a weld or connect constraint with the end of the rope should be straightforward. Did you try and fail?

Zcyyy, 2022-05-10:
1.Yes. I need a part of the rope pulled by a rigid body and the winch to lengthen the rope. When the length of the rope needs to be shortened, the winch rotates to pull the rope up, with the rigid body at the other end of the rope moving in a coordinated manner (the rope remains taut throughout the cycle of movement). I tried to do this with Mujoco so that I could show it in class and introduce Mujoco to teachers and other students.I also tried saving keyframes. After saving, the composite becomes a group of bodies. But they are still in the initial state, not the state I saved.
```
<mujoco model="Rope">
  <option timestep="0.002" jacobian="dense"/>
  <size nconmax="100" njmax="300" nstack="50000"/>
  <worldbody>
    <body name="box" pos="1.15 0 0.15" euler="0 0 0">
        <geom type="box" size="1 0.5 0.01" rgba="0 .7 0 1"/>
    </body>
    <body name="pulley" pos="0 0 0.15" euler="0 0 0">
        <geom type="cylinder" size="0.2 0.01" rgba="0 .7 0 1"/>
    </body>
    <body name="pulley2" pos="0 0 0.25" euler="0 0 0">
            <joint name="pin" type="hinge" axis="0 0 1" pos="0 0 0"/>
            <geom type="cylinder" size="0.1 0.1" rgba="0 .7 0 1"/>
            <body name="CB0" pos="0 0.1 -0.02" euler="0 0 0">
              <composite type="rope" count="50 1 1" spacing="0.023" offset="0 0 0" prefix="C">
                <joint kind="main" damping="0.005"/>
                <geom type="capsule" size=".005 .005" rgba=".8 .2 .1 1"/>
              </composite>
            </body>
    </body>
  </worldbody>
  <actuator>
          <velocity name="vel_servo" joint="pin" kv="10"/>
  </actuator>
</mujoco>  
```

2.I also have tried WELD and CONNECT, but when I use WELD or CONNECT connect body1 and body2, they do not float in their respective positions, although there are constraints between the two bodies.

yuvaltassa, 2022-05-17:
Hi @Zcyyy,

Here is a video of a modified model which I believe does what you want, including a demonstration of how to use keyframes.

https://user-images.githubusercontent.com/6387526/168820543-7b5abcdf-8bac-4015-90f6-fa9d39e7550e.mp4

Below is the modified model. You'll note I had to make quite a few tweaks to make it work. Unfortunately, ropes are quite tricky and difficult to model, and require some knowledge of the details of the simulator. I hope that by comparing your original model and mine, and tweaking some numbers, you'll be able to learn a bit about what makes such systems well-behaved.

```xml
<mujoco model="Rope">
  <option timestep="0.001" jacobian="dense"/>
  <size nconmax="500" njmax="1500" nkey="1"/>
  <default>
    <geom solref="1e-2 5" solimp=".5 .99 1e-3" friction=".5"/>
  </default>
  <worldbody>
    <light pos=".5 0 2"/>
    <geom type="box" pos="1.15 0 0.12" size="1 0.5 0.03" rgba="0 .7 0 1"/>
    <geom name="base" type="cylinder" pos="0 0 0.15" size="0.2 0.02" rgba="0 .7 0 1"/>
    <body name="pulley2" pos="0 0 0.25" euler="0 0 0">
      <joint name="pin" type="hinge" axis="0 0 1" pos="0 0 0" damping="1"/>
      <geom type="cylinder" size="0.1 0.1" rgba="0 .7 0 1"/>
      <body name="CB0" pos="0 0.1 -0.02" euler="0 0 0">
        <composite type="rope" count="50 1 1" spacing="0.023" offset="0 0 0" prefix="C">
          <joint kind="main" damping="0.0001" armature="0.00001"/>
          <geom type="capsule" size=".005 .01" rgba=".8 .2 .1 1"/>
        </composite>
      </body>
    </body>
    <body name="weight" pos="1.2 .1 .23">
      <freejoint/>
      <geom type="box" size=".05 .05 .05" rgba=".2 .2 .7 1"/>
    </body>
  </worldbody>
  <equality>
    <connect body1="weight" body2="CB49" anchor="-.05 0 0" solref="-10000 -1000"/>
  </equality>
  <actuator>
    <motor name="motor" joint="pin" gear="1" ctrllimited="true" ctrlrange="-3 3"/>
  </actuator>
</mujoco>  
```

YousefICSLAB, 2025-05-25:
[image]

Hi,

Building on your previous discussion, I’d like to extend your example to include an additional fixed pulley positioned between the main rotating pulley and the object, as illustrated in the attached figure.

Could you please assist me in implementing this setup? I intend to use two ropes:

The first rope should have its initial capsule connected to the rotating pulley, as in your original implementation.

The last capsule of the first rope should be connected to the first capsule of the second rope.

The final capsule of the second rope should then be connected to the object.

Does this approach make sense to you, and do you think it's a reasonable setup?