google-deepmind/mujoco #264: Reference frame - visual

danielstankw, 2022-05-08:
Hi everyone,
I am using Robosuite which is based on mujoco to perform RL.
The issue I have is that every time I try to visualize the coordinate system of robot joints the axis are thick and obscure the view. 
Can anyone suggest where can I find how to decrease the radius of those cylinders? 
[image]

yuvaltassa, 2022-05-08:
You'd want to tweak the `<visual> <scale/>...` [elements](https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-scale).

closed 2022-05-08 by yuvaltassa

danielstankw, 2022-05-08:
<mujoco model="ur5e">
    <actuator>
        
        
        <motor ctrllimited="true" ctrlrange="-150.0 150.0" joint="shoulder_pan_joint" name="torq_j1"/>
        <motor ctrllimited="true" ctrlrange="-150.0 150.0" joint="shoulder_lift_joint" name="torq_j2"/>
        <motor ctrllimited="true" ctrlrange="-150.0 150.0" joint="elbow_joint" name="torq_j3"/>
        <motor ctrllimited="true" ctrlrange="-28.0 28.0" joint="wrist_1_joint" name="torq_j4"/>
        <motor ctrllimited="true" ctrlrange="-28.0 28.0" joint="wrist_2_joint" name="torq_j5"/>
        <motor ctrllimited="true" ctrlrange="-28.0 28.0" joint="wrist_3_joint" name="torq_j6"/>
    </actuator>
    <asset>
        <mesh name="base" file="meshes/base.stl" />
        <mesh name="shoulder" file="meshes/shoulder.stl" />
        <mesh name="upperarm" file="meshes/upperarm.stl" />
        <mesh name="forearm" file="meshes/forearm.stl" />
        <mesh name="wrist1" file="meshes/wrist1.stl" />
        <mesh name="wrist2" file="meshes/wrist2.stl" />
        <mesh name="wrist3" file="meshes/wrist3.stl" />
        <mesh name="base_vis" file="meshes/base_vis.stl" />
        <mesh name="shoulder_vis" file="meshes/shoulder_vis.stl" />
        <mesh name="upperarm_vis" file="meshes/upperarm_vis.stl" />
        <mesh name="forearm_vis" file="meshes/forearm_vis.stl" />
        <mesh name="wrist1_vis" file="meshes/wrist1_vis.stl" />
        <mesh name="wrist2_vis" file="meshes/wrist2_vis.stl" />
        <mesh name="wrist3_vis" file="meshes/wrist3_vis.stl" />
    </asset>
    <worldbody>
        <body name="base" pos="0 0 0">
            
            <camera mode="fixed" name="robotview" pos="1.0 0 0.4" quat="0.653 0.271 0.271 0.653"/>
            <inertial diaginertia="0 0 0" mass="0" pos="0 0 0"/>
            
            <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.7 0.7 0.7 1" quat="0.707 0.707 0 0" mesh="base_vis" />
            <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="base" name="base_col"/>
            <body name="shoulder_link" pos="0 0 0.163">
                <inertial pos="0 0 0" mass="3.7" diaginertia="0.0102675 0.0102675 0.00666" />
                <joint name="shoulder_pan_joint" pos="0 0 0" axis="0 0 1" limited="true" range="-6.28319 6.28319" damping="0.001" frictionloss="0.01" />
                <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.7 0.7 0.7 1" quat="0.707 0.707 0 0" mesh="shoulder_vis" name="shoulder_visual" />
                <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="shoulder" name="shoulder_col" />
                <body name="upper_arm_link" pos="0 0.138 0" quat="0.707107 0 0.707107 0">
                    <inertial pos="0 0 0.2125" mass="8.393" diaginertia="0.133886 0.133886 0.0151074" />
                    <joint name="shoulder_lift_joint" pos="0 0 0" axis="0 1 0" limited="true" range="-6.28319 6.28319" damping="0.001" frictionloss="0.01" />
                    <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.7 0.7 0.7 1" quat="0.707 0.707 0 0" mesh="upperarm_vis" name="upperarm_visual" />
                    <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="upperarm" name="upperarm_col"/>
                    <body name="forearm_link" pos="0 -0.131 0.425">
                        <inertial pos="0 0 0.196" mass="2.275" diaginertia="0.0311796 0.0311796 0.004095" />
                        <joint name="elbow_joint" pos="0 0 0" axis="0 1 0" limited="true" range="-3.14159 3.14159" damping="0.001" frictionloss="0.01" />
                        <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.7 0.7 0.7 1" quat="0.707 0.707 0 0" mesh="forearm_vis" name="forearm_visual" />
                        <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="forearm" name="forearm_col"/>
                        <body name="wrist_1_link" pos="0 0 0.392" quat="0.707107 0 0.707107 0">
                            <inertial pos="0 0.127 0" mass="1.219" diaginertia="0.0025599 0.0025599 0.0021942" />
                            <joint name="wrist_1_joint" pos="0 0 0" axis="0 1 0" limited="true" range="-6.28319 6.28319" damping="0.001" frictionloss="0.01" />
                            <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.7 0.7 0.7 1" quat="0.707 0.707 0 0" mesh="wrist1_vis" name="wrist1_visual" />
                            <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="wrist1" name="wrist1_col" />
                            <body name="wrist_2_link" pos="0 0.127 0">
                                <inertial pos="0 0 0.1" mass="1.219" diaginertia="0.0025599 0.0025599 0.0021942" />
                                <joint name="wrist_2_joint" pos="0 0 0" axis="0 0 1" limited="true" range="-6.28319 6.28319" damping="0.001" frictionloss="0.01" />
                                <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.7 0.7 0.7 1" quat="0.707 0.707 0 0" mesh="wrist2_vis" name="wrist2_visual" />
                                <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="wrist2" name="wrist2_col" />
                                <body name="wrist_3_link" pos="0 0 0.1">
                                    <inertial pos="0 0.0771683 0" quat="0.707107 0 0 0.707107" mass="0.1889" diaginertia="0.000132134 9.90863e-05 9.90863e-05" />
                                    <joint name="wrist_3_joint" pos="0 0 0" axis="0 1 0" limited="true" range="-6.28319 6.28319" damping="0.001" frictionloss="0.01" />
[... cut ...]
                                    <body name="right_hand" pos="0 0.098 0" quat="0.707 -0.707 0 0" >
                                        
                                        <camera mode="fixed" name="eye_in_hand" pos="0.05 0 0" quat="0 0.707108 0.707108 0" fovy="75"/>
                                        
                                    </body>
                                </body>
                            </body>
                        </body>
                    </body>
                </body>
            </body>
        </body>
    </worldbody>
</mujoco>

danielstankw, 2022-05-08:
```
<mujoco model="ur5e">
    <actuator>
        <!-- Physical limits of the actuator. -->
        <!-- Values taken from https://www.universal-robots.com/articles/ur-articles/max-joint-torques/ -->
        <motor ctrllimited="true" ctrlrange="-150.0 150.0" joint="shoulder_pan_joint" name="torq_j1"/>
        <motor ctrllimited="true" ctrlrange="-150.0 150.0" joint="shoulder_lift_joint" name="torq_j2"/>
        <motor ctrllimited="true" ctrlrange="-150.0 150.0" joint="elbow_joint" name="torq_j3"/>
        <motor ctrllimited="true" ctrlrange="-28.0 28.0" joint="wrist_1_joint" name="torq_j4"/>
        <motor ctrllimited="true" ctrlrange="-28.0 28.0" joint="wrist_2_joint" name="torq_j5"/>
        <motor ctrllimited="true" ctrlrange="-28.0 28.0" joint="wrist_3_joint" name="torq_j6"/>
    </actuator>
    <asset>
        <mesh name="base" file="meshes/base.stl" />
        <mesh name="shoulder" file="meshes/shoulder.stl" />
        <mesh name="upperarm" file="meshes/upperarm.stl" />
        <mesh name="forearm" file="meshes/forearm.stl" />
        <mesh name="wrist1" file="meshes/wrist1.stl" />
        <mesh name="wrist2" file="meshes/wrist2.stl" />
        <mesh name="wrist3" file="meshes/wrist3.stl" />
        <mesh name="base_vis" file="meshes/base_vis.stl" />
        <mesh name="shoulder_vis" file="meshes/shoulder_vis.stl" />
        <mesh name="upperarm_vis" file="meshes/upperarm_vis.stl" />
        <mesh name="forearm_vis" file="meshes/forearm_vis.stl" />
        <mesh name="wrist1_vis" file="meshes/wrist1_vis.stl" />
        <mesh name="wrist2_vis" file="meshes/wrist2_vis.stl" />
        <mesh name="wrist3_vis" file="meshes/wrist3_vis.stl" />
    </asset>
    <worldbody>
        <body name="base" pos="0 0 0">
            <!-- robot view -->
            <camera mode="fixed" name="robotview" pos="1.0 0 0.4" quat="0.653 0.271 0.271 0.653"/>
            <inertial diaginertia="0 0 0" mass="0" pos="0 0 0"/>
            <!-- mount attached here -->
            <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.7 0.7 0.7 1" quat="0.707 0.707 0 0" mesh="base_vis" />
            <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="base" name="base_col"/>
            <body name="shoulder_link" pos="0 0 0.163">
                <inertial pos="0 0 0" mass="3.7" diaginertia="0.0102675 0.0102675 0.00666" />
                <joint name="shoulder_pan_joint" pos="0 0 0" axis="0 0 1" limited="true" range="-6.28319 6.28319" damping="0.001" frictionloss="0.01" />
                <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.7 0.7 0.7 1" quat="0.707 0.707 0 0" mesh="shoulder_vis" name="shoulder_visual" />
                <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="shoulder" name="shoulder_col" />
                <body name="upper_arm_link" pos="0 0.138 0" quat="0.707107 0 0.707107 0">
                    <inertial pos="0 0 0.2125" mass="8.393" diaginertia="0.133886 0.133886 0.0151074" />
                    <joint name="shoulder_lift_joint" pos="0 0 0" axis="0 1 0" limited="true" range="-6.28319 6.28319" damping="0.001" frictionloss="0.01" />
                    <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.7 0.7 0.7 1" quat="0.707 0.707 0 0" mesh="upperarm_vis" name="upperarm_visual" />
                    <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="upperarm" name="upperarm_col"/>
                    <body name="forearm_link" pos="0 -0.131 0.425">
                        <inertial pos="0 0 0.196" mass="2.275" diaginertia="0.0311796 0.0311796 0.004095" />
                        <joint name="elbow_joint" pos="0 0 0" axis="0 1 0" limited="true" range="-3.14159 3.14159" damping="0.001" frictionloss="0.01" />
                        <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.7 0.7 0.7 1" quat="0.707 0.707 0 0" mesh="forearm_vis" name="forearm_visual" />
                        <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="forearm" name="forearm_col"/>
                        <body name="wrist_1_link" pos="0 0 0.392" quat="0.707107 0 0.707107 0">
                            <inertial pos="0 0.127 0" mass="1.219" diaginertia="0.0025599 0.0025599 0.0021942" />
                            <joint name="wrist_1_joint" pos="0 0 0" axis="0 1 0" limited="true" range="-6.28319 6.28319" damping="0.001" frictionloss="0.01" />
                            <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.7 0.7 0.7 1" quat="0.707 0.707 0 0" mesh="wrist1_vis" name="wrist1_visual" />
                            <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="wrist1" name="wrist1_col" />
                            <body name="wrist_2_link" pos="0 0.127 0">
                                <inertial pos="0 0 0.1" mass="1.219" diaginertia="0.0025599 0.0025599 0.0021942" />
                                <joint name="wrist_2_joint" pos="0 0 0" axis="0 0 1" limited="true" range="-6.28319 6.28319" damping="0.001" frictionloss="0.01" />
                                <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.7 0.7 0.7 1" quat="0.707 0.707 0 0" mesh="wrist2_vis" name="wrist2_visual" />
                                <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="wrist2" name="wrist2_col" />
                                <body name="wrist_3_link" pos="0 0 0.1">
                                    <inertial pos="0 0.0771683 0" quat="0.707107 0 0 0.707107" mass="0.1889" diaginertia="0.000132134 9.90863e-05 9.90863e-05" />
[... cut ...]
                                        <!-- This camera points out from the eef. -->
                                        <camera mode="fixed" name="eye_in_hand" pos="0.05 0 0" quat="0 0.707108 0.707108 0" fovy="75"/>
                                        <!-- To add gripper -->
                                    </body>
                                </body>
                            </body>
                        </body>
                    </body>
                </body>
            </body>
        </body>
    </worldbody>
</mujoco>
```

danielstankw, 2022-05-08:
I dont have any of those parameters of XML that you mentioned

yuvaltassa, 2022-05-08:
No, indeed you do not have them. This is why I am proposing that you _add_ them.

[Here](https://github.com/deepmind/mujoco/blob/main/model/mug/mug.xml) for example is an XML with a `visual` clause.

danielstankw, 2022-05-10:
> No, indeed you do not have them. I am proposing that you _add_ them.
> 
> [Here](https://github.com/deepmind/mujoco/blob/main/model/mug/mug.xml) for example is an XML with a `visual` clause.

Yes, but adding them didnt help as when i run the simulation they are already there. I just need to find how to modify them which I am havingissues with.

yuvaltassa, 2022-05-10:
What do you mean "they are already there"? You mean you want to modify visual scales at runtime, after the model has been compiled?

If you are using our python bindings via [dm_control](https://github.com/deepmind/dm_control), given a compiled `model`, you would do something like

```py
model.vis.scale.framewidth = x
```

I hope this is what you mean.

kevinzakka, 2022-05-10:
Chiming in with [an example from dm_control]( https://github.com/deepmind/dm_control/blob/4e1a35595124742015ae0c7a829e099a5aa100f5/dm_control/composer/arena.xml#L9).