google-deepmind/mujoco #266: [Bug] `multiccd` option not

kevinzakka, 2022-05-10:
Tried [`multiccd`](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-flag) for fun and got an error on both the MuJoCo app (2.1.5) and PyMJCF from dm_control:

```python
AttributeError: Line 12: error while parsing element <option>: 'multiccd' is not a valid attribute for <option>. # PyMJCF.
XML Error: Schema violation: unrecognized attribute: 'multiccd'. # App.
```

kevinzakka, 2022-05-10:
Apologies, I realize this should have been under the flag attribute.

```xml
<option>
  <flag multiccd="enable"/>
</option>
```

closed 2022-05-10 by kevinzakka

yuvaltassa, 2022-05-10:
Actually, while we're here, there is a bug (fixed internally, to be pushed soon): The multiCCD disable/enable button in `simulate` does not work.