google-deepmind/mujoco #269: "Error: engine error: Missing or invalid license activation"

mjuric2, 2022-05-10:
Hello,
I am getting this error ("Error: engine error: Missing or invalid license activation") while trying to read lines:

    mjModel * m = (mjModel *) malloc(sizeof(mjModel));
    m = mj_loadXML("/home/marko/ZR/src/humanoid.xml", NULL, errstr, 128);

I assume it is not recgnising mjkey.txt file but i have linked that directory in CMakeLists.txt.
Here it is:

cmake_minimum_required(VERSION 3.10)

# set the project name and version
project(ZR VERSION 1.0)
project(ZR)

# specify the C++ standard
set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED True)
set(CMAKE_CXX_FLAGS -pthread)

set(MUJOCO_PATH /home/marko/mujoco/mujoco200_linux)
link_directories(${MUJOCO_PATH}/bin)

add_executable(start start.cpp)

find_package(GLEW REQUIRED)
find_package(Threads REQUIRED)

target_include_directories(start PUBLIC ${MUJOCO_PATH}/include)
target_link_libraries(start PUBLIC libglew.so libglewegl.so libglewosmesa.so libglfw.so.3 libmujoco200nogl.so)

I'd be glad if you could help me.

kevinzakka, 2022-05-10:
Out of curiosity, why are you using the license key still?

mjuric2, 2022-05-11:
There is no, is your advice to install a newest version?

saran-t, 2022-05-11:
Unless you have a very specific reason for using version 2.0.0, I'd recommend using the latest one (currently 2.1.5) instead.