google-deepmind/mujoco #270: How to reset the environment to a specific state in python

fanyangcmu, 2022-05-10:
I am trying to reset the mujoco environment to a specific state. Here is what I have tried:
I would first record the state I want to reset to with
`reset_state = env.sim.get_state()`
Then I would call
`def set_state(env, reset_state):
    env.sim.set_state(reset_state)
    env.sim.forward()`
to reset the current environment to the desired state. 
If I input the same action to the environment after the resetting, this does not necessarily reproduce the trajectory. Specifically, if the agent does not collide with any obstacle in my environment, I can successfully reproduce the results. But once the collision happens, where would be a slight difference in the state, though if I render the simulation, it is visually the same. 
Does anyone know why this happens? I wonder if there is any randomness in collision simulation for Mujoco. Thanks

kevinzakka, 2022-05-10:
Have you disabled the [warmstart mechanism](https://mujoco.readthedocs.io/en/latest/XMLreference.html?highlight=warmstart#option-flag)?

kevinzakka, 2022-05-12:
@fanyangcmu When you say reset what do you mean? Are you saving it when recording the sim state and then restoring it when playing back?

yuvaltassa, 2022-05-14:
To confirm, MuJoCo has no hidden states and is completely deterministic.

@kevinzakka's advice is correct.

closed 2022-05-14 by yuvaltassa

fanyangcmu, 2022-05-18:
@kevinzakka Yes, exactly. Thanks for pointing out that Mujoco is completely deterministic. I have done the following experiments and have the following results.
1. I use a specific sequence of action to rollout an episode, and record the state I want to reset to. This trajectory is referred to as trajectory A for convenience. 
2. I reset the environment to the state I want to reset to. And I use the same action to rollout the trajectory. This trajectory is referred to as trajectory B.
3. I reset again to the state I want to reset to and use the same action to rollout again. This trajectory is referred to as trajectory C.

Here is what I have observed:
1. trajectory B is the same as trajectory C
2. Trajectory A is the same as trajectory B until any collision happens. 

If there is nothing wrong with the hidden state and the warm start part, I think the issue may be within the specific environment I am using.
Thank you for your help! I appreciate it a lot.

yuvaltassa, 2022-05-18:
Hi,

To clarify: if warmstarts are not disabled then `qacc_warmstart` **is a part of the state** if you want perfect reproducibility. If you saved the *full* state and controls, you should always get perfect reproducibility, collisions should not have any effect.

If what I said above appears not to hold for you, then we would love to see a MWE demonstrating this.