google-deepmind/mujoco #271: How to compute the gradient of reward w.r.t. torso mass?

ZJP-c4cld, 2022-05-11:
I want to construct a neural network in pytorch which outputs appropriate values of mujoco model parameters. But this requires that mujoco model parameters are converted into pytorch tensors , and mujoco environments run well with these converted model parameters, and compute the gradient of reward w.r.t the mujoco model parameters. 
Anyone know how to achieve this goal?

yuvaltassa, 2022-05-14:
Hi, 

It is not clear what you are trying to do. Can you articulate yourself more clearly, ideally with a specific example of what your desired achitechture would accomplish?

ZJP-c4cld, 2022-05-14:
@yuvaltassa Thank you for your reply. 

My research interest is robustness of RL algorithms to environment parameters. I want to modify currents RL algorithms to make them achieve good performance when they are tested in environments with unfamilar parameters. (For example, an agent is trained in Cartpole environment with 1m pole. I want it achieve good performance in Cartpole environment with 3m pole.)

  To achieve this goal, I want to get the relationship between model parameter values and RL algorithm's performance (reward). As a result, I want to get the gradient of reward with respect to the model parameters.

Mujoco simulator has applied in my experiments. But Mujoco simulator is not implemented by pure python. So I cannot get the the gradient of reward with respect to the model parameters. 

 So my queation is:
 Can Mujoco Python Binding compute the gradient of reward with respect to the model parameters? If so, how should I use it to achieve this goal?

yuvaltassa, 2022-05-14:
Thanks for the explanation. 

You cannot compute this gradient exactly as the learning process is non-deterministic and the gradient you are looking for is not well-defined. However, just like in most RL algorithms which approximate the gradient of the reward with respect to network parameters (see e.g., [here](https://www.jmlr.org/papers/v5/greensmith04a.html)), you can approximate your desired gradient in the same way.

In order to introduce variation into MuJoCo models you can either procedurally modify the input XML yourself, or use the [composer/variation](https://github.com/deepmind/dm_control/tree/main/dm_control/composer/variation) module of `dm_control`. If you want to use this module and need help, I suggest that you ask in the `dm_control` [GitHub repo](https://github.com/deepmind/dm_control).

closed 2022-05-14 by yuvaltassa