google-deepmind/mujoco #272: Support for 32-bit Python on Windows?

1cookspe, 2022-05-11:
Hello,

We are trying to install Mujoco into our project using 32-bit Python. However, we cannot install it using `pip install mujoco` or through our virtual environment in PyCharm. It has worked perfectly for us in 64-bit Python, however we must use 32-bit Python due to our code's dependence on another library that only works on 32-bit Python.

We notice that the releases page only lists the Mujoco packages for 64-bit Windows. Is there a way for us to install the Python bindings for 32-bit Python?

Thank you!

saran-t, 2022-05-11:
We haven't released a 32-bit Windows binary because so far no one's asked of it, and we'd prefer to not release a large number of different binaries that we need to support.

Having said that, we're getting very close to open sourcing the library so when that happens you'll be able to build a 32-bit binary yourself if required.

1cookspe, 2022-05-11:
@saran-t Thanks for the response! We have decided to build mujoco for our 32 bit Python from source using the instructions in the [Python README](https://github.com/deepmind/mujoco/blob/main/python/README.md). 

However, we are also facing several errors when completing this. For one, when we run the `make_sdist.sh` script, we receive the following error:
`ModuleNotFoundError: No module named 'introspect'`.
However, we have the 'introspect' folder (as we pulled this repository from GitHub). Despite this error, the `dist` folder is still created.

Then, we set our MUJOCO_PATH like so `set MUJOCO_PATH=C:\Users\username\mujoco`, where `mujoco` is the folder that we downloaded the binary from the [GitHub releases page](https://github.com/deepmind/mujoco/releases). We then run `pip install mujoco-2.1.5.tar.gz`, and it presents us with this following error:

```
running bdist_wheel
      running build
      running build_py
      creating build
      creating build\lib.win32-cpython-39
      creating build\lib.win32-cpython-39\mujoco
      copying mujoco\bindings_test.py -> build\lib.win32-cpython-39\mujoco
      copying mujoco\render_test.py -> build\lib.win32-cpython-39\mujoco
      copying mujoco\rollout.py -> build\lib.win32-cpython-39\mujoco
      copying mujoco\rollout_test.py -> build\lib.win32-cpython-39\mujoco
      copying mujoco\__init__.py -> build\lib.win32-cpython-39\mujoco
      creating build\lib.win32-cpython-39\mujoco\egl
      copying mujoco\egl\egl_ext.py -> build\lib.win32-cpython-39\mujoco\egl
      copying mujoco\egl\__init__.py -> build\lib.win32-cpython-39\mujoco\egl
      creating build\lib.win32-cpython-39\mujoco\glfw
      copying mujoco\glfw\__init__.py -> build\lib.win32-cpython-39\mujoco\glfw
      creating build\lib.win32-cpython-39\mujoco\osmesa
      copying mujoco\osmesa\__init__.py -> build\lib.win32-cpython-39\mujoco\osmesa
      running build_ext
      error: [WinError 2] The system cannot find the file specified
```

Any ideas if we're doing anything wrong? We are setting `MUJOCO_PATH` to the base mujoco folder of the binary installation, however please let us know if this is not correct.

kevinzakka, 2022-05-11:
@1cookspe do you need to `set` in the command?

1cookspe, 2022-05-11:
@kevinzakka Thanks for your response. I am not sure if `set` is required. However, we have checked that the `MUJOCO_PATH` was set to our intended folder through `echo %MUJOCO_PATH%`, and it printed out the folder where we saved our Mujoco installation.

saran-t, 2022-05-11:
I haven't had a chance to look into your issue with `introspect` in detail yet, but even if you manage to build the Python bindings for 32-bit, you'd still not be able to use it since the underlying MuJoCo library isn't available for 32-bit.

1cookspe, 2022-05-11:
@saran-t Ok thanks for the input!

kevinzakka, 2022-05-11:
@1cookspe Interesting, `set` on my mac does not correctly set the variable.

saran-t, 2022-05-11:
@1cookspe Can you try using `/` instead of `\` in `MUJOCO_PATH`? Also, what shell are you using? We only test against Cygwin so it's likely that the script doesn't work correctly in other setup.

1cookspe, 2022-05-19:
Hi! Sorry for the late response! Using `/` instead of `\` worked; thank you so much!

closed 2022-05-19 by 1cookspe