google-deepmind/mujoco #273: How to import new urdf models in mujoco?

soumy97, 2022-05-11:
Hello,
I have a urdf file of an inverted pendulum which I want to import in mujoco but I am not able to find any details regarding that. In other platforms, I was able to have functions to do the importing but I don't have anything in mujoco. Please help me out.

kevinzakka, 2022-05-11:
This thread from the old forum might prove useful: https://roboti.us/forum/index.php?threads/meshes-ignored-when-converting-urdf-to-mjcf.3433/

soumy97, 2022-05-23:
Hello,
      I tried to do as you said but this error is showing up whenever I am dragging and dropping the urdf file in the simulate app. 
   **Error: could not open STL file '/home/soumyajit/Downloads/Inverted/urdf/base_link.STL' 
   Object name = base_link, id = 0**
Also, I have attached the Urdf file for your reference.
[Inverted.zip](https://github.com/deepmind/mujoco/files/8757107/Inverted.zip)

v-r-a, 2022-05-23:
I too face the same problem while using the outputs of solidworks2urdf or fusion2urdf scripts.

I am not sure how the file is parsed but it is clear that there is no mesh file at the said path: /home/soumyajit/Downloads/Inverted/urdf

I just copy all the meshes (.stl files) to the /urdf directory and it works fine. (It is working for your example too).

It would be great if someone throws light on the details of why this happens.

v-r-a, 2022-05-23:
> This thread from the old forum might prove useful: https://roboti.us/forum/index.php?threads/meshes-ignored-when-converting-urdf-to-mjcf.3433/

(a) Unfortunately, examples in this forum cannot be downloaded anymore. Can we do so, from somewhere else?
(b) The solutions include adding Mujoco tag (and specifying the mesh path) in the urdf file. Can we do so without adding the Mujoco tag?

yuvaltassa, 2022-05-23:
@v-r-a Why can't you add a `<mujoco>` tag? If you don't add that tag, how do you expect to tell MuJoCo where the meshes are? And it looks like already have the workaround of just copying all the meshes to the same directory, so you can avoid using the `<mujoco>` tag if you really want. So basically it's not clear to me what the issue is.

@soumy97 please add a `<mujoco>` tag, as documented [here](https://mujoco.readthedocs.io/en/latest/modeling.html?highlight=urdf#urdf-extensions).

closed 2022-05-23 by yuvaltassa

v-r-a, 2022-05-23:
> @v-r-a Why can't you add a `<mujoco>` tag? If you don't add that tag, how do you expect to tell MuJoCo where the meshes are? And it looks like already have the workaround of just copying all the [...]

(1) I was just wondering if the path that is specified already in urdf could be used. For example: 
<geometry>
        <mesh
          filename="package://Inverted/meshes/base_link.STL" />
</geometry>

(2) This is a query for another day, but are there any plans to implement inverse kinematics in MuJoCo? I was trying to avoid  <mujoco> tag to maintain compatibility with a library that can do inverse kinematics, e.g., https://github.com/rbdl/rbdl .

yuvaltassa, 2022-05-23:
1. This is certainly possible. You are welcome to submit a pull request 😄. 
2. No such plans. There are many ways to implement IK and implementing just one of them doesn't make sense, there will always be someone unhappy with the implementation. However, given analytic Jacobians (which MuJoCo computes very efficiently) any one of these implementations is rather easy to write. [Here](https://github.com/deepmind/dm_control/blob/main/dm_control/utils/inverse_kinematics.py) is an example from DeepMind.

wangcongrobot, 2022-06-01:
Hi @yuvaltassa. I wrote a simple [example](https://github.com/wangcongrobot/dual_ur5_husky_mujoco) to convert the urdf to mujoco xml. While in the new version(2.2), I don't find the **_compile_** in **_bin_** folder. So how could we deal with it now?

kevinzakka, 2022-06-01:
@wangcongrobot Unsure about why compile isn't part of the binaries anymore but you can load the XML file in the simulate app and it will compile it for you.