google-deepmind/mujoco #279: How to move the whole model?

ycntzx, 2022-05-17:
I have designed a simple model and I want it to be able to move upwards as a whole to touch an obstacle. How do I achieve this? THX!
Here is my simple model:

<mujoco model="roboteri">
    <option timestep="0.002" iterations="50" apirate="1000" gravity="disable" solver="Newton" tolerance="1e-10" cone="elliptic"/>
    <compiler angle="degree" meshdir="meshes" balanceinertia="true"/>

    <default>
        <light castshadow="false" diffuse="0 0 0" />
        <camera fovy="60" />
        <site type="sphere" rgba="1 0 0 1" size="0.4"/>
    </default>

    <asset>

        <mesh file="ManipulatorDisk_start.STL" />
        <mesh file="ManipulatorDisk_grob.STL"/>
        <mesh file="ManipulatorEndDisk.STL"/>

        <texture name="floor_tile" file="meshes/floor_tile.png" width="3000" height="3000" type="2d"/>
        <material name="tile" texture="floor_tile" specular="0" shininess="0" reflectance="0" emission="1" />

        <material name="carbon" specular="1" shininess="1" reflectance="1" rgba="0.82 0.41 0.12 1" emission="1" />
        <texture name="foam_blocks" file="meshes/foam_blocks.png" width="2744" height="2744" type="2d"/>
        <material name="foam" texture="foam_blocks" specular="0" shininess="0" reflectance="0" emission="1" />
    </asset>

    <default>

		<default class="G">
                        <joint type="hinge" limited="true" range="-28.12 28.12" springdamper="0.005 10"/>
                        </default>
		<default class="shoko">
						<geom type="mesh" pos="15 0 0"  euler="90 -90 0" rgba="1 0.5 0.31 1"  mesh="ManipulatorDisk_grob"/>
						</default>
		<default class="blue">
						<geom type="mesh" rgba="0 .31 .61 1" pos="0 0 0"  euler="90 0 0"  mesh="ManipulatorDisk_grob"/>
						</default>
        <default class="red">
						<geom type="mesh" rgba="1 0 0 1" pos="15 0 0"  euler="90 -90 0"  mesh="ManipulatorEndDisk"/>
						</default>

	 </default>

    <worldbody>
        <geom name="floor_foam_block" pos="0 0 0" size="100 100 0.1" type="plane" rgba="1 0.83 0.61 1" material="foam"/>
        <geom name="floor_grey_tile" pos="0 0 -0.001" size="120 120 0.1" type="plane" material="tile"/>
        <light directional="true" pos="-0.5 0.5 3" dir="0 0 -1" />

        <body name="base_Hand" pos="0 0 0">
            <geom name="ManipulatorDisk_start" type="mesh" mesh="ManipulatorDisk_start"  pos="0 0 0" euler="90 0 0" material="carbon" />
            <site name="b0" pos="2.49 -5.23 46"/> <site name="b00" pos="11.39 -3.61 46"/> <site name="b000" pos="9.77 -12.57 46"/>

            <body name="disk1" pos="0 0 48.05">
                 <inertial pos="0 0 0" mass="0.00001" diaginertia="0.3 0.3 0.3"/>
                 <geom name="g1" class="shoko" />
                 <joint name="joint1" pos="0 -7.5 -0.75" class="G" axis="1 0 0"/>
                 <site name="b1" pos="2.49 -5.23 0.9"/> <site name="B1" pos="2.49 -5.23 3.9"/>
                 <site name="b11" pos="11.39 -3.61 1"/> <site name="B11" pos="11.39 -3.61 4"/>
                 <site name="b111" pos="9.77 -12.57 1.2"/> <site name="B111" pos="9.77 -12.57 4.2"/>

                 <body name="disk2" pos="0 0 5.738">
                    <inertial pos="0 0 0" mass="0.00001" diaginertia="0.3 0.3 0.3"/>
                    <geom name="g2" class="blue"/>
                    <joint name="joint2" pos="7.5 0 -0.75" class="G"  axis="0 1 0"/>
                    <site name="b2" pos="2.49 -5.23 1.2"/> <site name="B2" pos="2.49 -5.23 4.2"/>
                    <site name="b22" pos="11.39 -3.61 1"/> <site name="B22" pos="11.39 -3.61 4"/>
                    <site name="b222" pos="9.77 -12.57 0.9"/> <site name="B222" pos="9.77 -12.57 3.9"/>

                    <body name="disk3" pos="0 0 5.738">
[... 130 lines cut ...]
            <site site="b333" />
            <site site="B333" />
            <site site="b444" />
            <site site="B444" />
            <site site="b555" />
            <site site="B555" />
            <site site="b666" />
            <site site="B666" />
            <site site="b777" />
            <site site="B777" />
            <site site="b888" />
            <site site="B888" />
            <site site="b999" />

        </spatial>
    </tendon>

    <actuator>
    <motor    name="s1" ctrllimited="true" ctrlrange="-100 0" tendon="t1" gear="100"/>
    <motor    name="s2" ctrllimited="true" ctrlrange="-100 0" tendon="t2" gear="100"/>
    <motor    name="s3" ctrllimited="true" ctrlrange="-100 0" tendon="t3" gear="100"/>

    </actuator>

</mujoco>

yuvaltassa, 2022-05-17:
When you say "the whole model", you presumably mean that you want to move the base body "base_Hand"?

You can add a vertical slider joint to this body:
```xml
<body name="base_Hand" pos="0 0 0">
  <joint type="slide" axis="0 0 1"/>
  ...
```

closed 2022-05-17 by yuvaltassa

ycntzx, 2022-05-17:
@yuvaltassa Thanks for the quick reply, but this actuator didn't work. This is so strange, all my other actuator settings work.